lvgl/lv_demos #4: lv_tutorial_keyboard: emulated keyboard not working when message box is open

StephanSy, 2018-06-14:
When the message box is opened the emulated keyboard is disabled, which does not really make sense.

kisvegabor, 2018-06-14:
Hi,

If you mean you can't press the Next/inc/Dec/Enter button when the message box is shown then it works for me. See the image. Or am I misunderstood something?

[image]

StephanSy, 2018-06-15:
Sorry, I only pressed the next button and expected that it would switch between the yes and no button. But the LV_GROUP_KEY_NEXT key is 
- firstly not sent by the group to the message box because of the focus freeze and 
- secondly, the message box does not use this signal to switch between buttons as I was expecting. 

Wouldn't it make sense that the  LV_GROUP_KEY_NEXT/LV_GROUP_KEY_PREV keys are also sent to the object in focus? An even when the focus is frozen?

An that the next key also switches between the buttons in the message box?

BR,
Stephan

StephanSy, 2018-06-15:
Or on second thought: the LV_GROUP_KEY_NEXT/LV_GROUP_KEY_PREV keys should be sent to the object if and only if the focus is frozen?

kisvegabor, 2018-06-17:
The LV_GROUP_KEY_NEXT/LV_GROUP_KEY_PREV is uses to focus on the next/previous object. In this case the message box is logically one object. So once you focused on it you can "move" in it with LV_GROUP_KEY/UP/DOWN/LEFT/RIGHT. Maybe renaming of LV_GROUP_KEY_NEXT/PREV to LV_GROUP_KEY_**FOCUS**_NEXT/PREV would make it clearer. 

Freeze is used to prevent focusing on the next object to make easier the use of modal dialogues.

StephanSy, 2018-06-18:
I see, but when with the "next" button/key the focus switches between buttons, the user will expect that this also works the same way within the message box.

In my special case, a rotary switch with push button is the only input device I have. So depending on the "edit" state the GUI, I need to interpret keys as NEXT/PREV or as LEFT/RIGHT/.

BR
Stephan

kisvegabor, 2018-06-19:
I understand it, but is wouldn't be logical because it's still one object with "sub-objects". 
Imagine similar but more complex objects like a keyboard (~35 buttons). It's an object, if you focus on it you can navigate on it. 
In my opinion it just wouldn't be logical to distinguish objects this way. If you really want this with a message box then you should create it from a container (which can _fit_ to the content is you want) and a few button and add the buttons  to the group directly.

As mentioned [this issue](https://github.com/littlevgl/lvgl/issues/277) a better encoder support is added. I hope it will make easier to handle the default message box with an encoder using it.

embeddedt, 2018-12-10:
There was no activity here for nearly 6 months so I am going to close this issue.

If you wish to continue the discussion feel free to comment here and reopen the issue.

closed 2018-12-10 by embeddedt