lvgl/lv_demos #11: keyboard tutorial crashes when Yes or No is pressed

rbultman, 2018-10-12:
This seems to be related to deleting the mbox in `mbox_action`.

I'm running this under mbed but not multi-threading.
I'm using the NUCLEO-F446RE dev board with [this display](https://www.ebay.com/itm/3-2-inch-TFT-LCD-Module-Display-Power-than-SainSmart-w-Capacitive-Touch-Tutorial/291870027140?hash=item43f4cf2d84:g:B~gAAOSwsiZbRc5t:rk:18:pf:0) connected in 16-bit parallel.
`lv_tic_inc` and `lv_task_handler` are called from the main loop context, not from a timer.
I can get it to not crash if I hide the mbox instead of deleting it in `mbox_action`.

In `mbox_action`, if I call 
```
lv_obj_set_hidden(lv_obj_get_parent(mbox), true); 
```
instead of 
```
lv_obj_del(lv_obj_get_parent(mbox));
```
The app does not crash.
If I create a variable to save the mbox on creation
```
static lv_obj_t * mbox = NULL;
```
Then I can change `enable_action` to delete the 'old' mbox and create a new one.  This works fine.
```
static lv_res_t enable_action(lv_obj_t * btn)
{
   /*If the butto nsi released the show message box to be sure about the Enable*/
   if(lv_btn_get_state(btn) == LV_BTN_STATE_REL) {
      if (mbox != NULL) {
         printf("mbox existed previously, deleting prior to creating new.\r\n");
         lv_obj_del(lv_obj_get_parent(mbox));    /*Delete the black background. (it will delete the mbox too)*/
      }
...
      /*Create a message box*/
      //lv_obj_t * mbox = lv_mbox_create(bg, NULL);
      mbox = lv_mbox_create(bg, NULL);
```

Based on the crash report and the memory map, it seems to be crashing in memcpy after the object is deleted.

Is it possible that somehow a touch is trying to be delivered to the deleted mbox and deleting it later is ok?

kisvegabor, 2018-10-12:
Hi,

Are you using the latest version form the `master` branch?

rbultman, 2018-10-15:
The register dump included the program counter.  The address from the program counter falls in the address range occupied by memcpy in the memory map.  mbed just dumps the register contents to the UART along with a cryptic message about fault exception.

I'm just using mbed with messaging printed on the UART.  I'll see if I can get the project running in Eclipse and get a full stack trace.  I'll try your (3) first though.

kisvegabor, 2018-10-15:
Great, thank you!

rbultman, 2018-10-16:
@kisvegabor I did your (3) and it works, i.e. it does not crash.  The inc/dec buttons no longer work, but I guess that was expected because they were part of the now non-existent group.

I'm still working on (2).  I'm trying get the build working on a Windows computer.  This might take a little more time.

kisvegabor, 2018-10-16:
Thank you for the feedback.

Yesterday a few group related issues were revealed and fixed in the `master` branch. It's very probable that it will solve this issue too. Can you test again with the latest `master`?

rbultman, 2018-10-16:
Will do.

rbultman, 2018-10-16:
I updated to latest master and the problem went away.  Thanks for the update.

closed 2018-10-16 by rbultman

kisvegabor, 2018-10-16:
Great!

Thank you very much for the test and the feedback!