lvgl/lv_demos #12: Releases

seyyah, 2018-10-15:
TODO: There is not releases same as https://github.com/littlevgl/lvgl/releases

kisvegabor, 2018-10-16:
There are no releases in the "lv_examples" right now because it just mainly follows the changes of "lvgl" and very rarely gets really new updates (like new apps).

seyyah, 2018-10-17:
Because we have to old version (v5.1.1), we need the compatible version for lv_examples.

When i examined the lv_projects, I think that lv_examples versions can be like in table. If it's okay for you, i tag the releases.

| version | commit |
|-----------|------------|
| v5.1.2  | 4f9f7510b181e739cbc70d2f6b56ea657ee61967 |
| v5.1.1 | 6e005d978c0323426943d25e25082be922958506 |
| v5.1.0 | d6b40a26fd59be56e730be8be25bc4b8db8e61ef |
| v5.0.2 | 154252c24d1f31b82c27be9233e62f6b7daf91b0 |
| v5.0.1 | 0174d34e4c41ab8e0ea60f3aead02eb9145b2426 |

For example,

[image]

- lvgl v5.1.1: https://github.com/littlevgl/lvgl/tree/522d20d9e4f896300503b3cfe744dfff4aff4aa1
- lv_examples: https://github.com/littlevgl/lv_examples/tree/6e005d978c0323426943d25e25082be922958506

So, i think that `6e005d978c0323426943d25e25082be922958506` commit for v5.1.1

seyyah, 2018-10-17:
I create Github releases via console,

[image]

But I can't pull request. So can you add a colloborator for https://github.com/littlevgl/lv_examples?

seyyah, 2018-10-17:
Invitation didn't come. Can you send invitation link?

kisvegabor, 2018-10-18:
I sent the link in email

seyyah, 2018-10-18:
It's ready

[image]

closed 2018-10-18 by seyyah

kisvegabor, 2018-10-19:
Awesome! :)