lvgl/lv_demos #26: [request] "Settings menu" demo

puzrin, 2019-03-26:
A lot of small embedded devices have interface like this:

1. Main screen
2. Settings menu with tons of params (mostly numeric, sometime text)

IMHO it would save a lot of time, if everyone could take ready example, and copy-paste code with controls of required type.

What is expected from menu:

- Good design
- Proper interaction with different input devices (touch/buttons)
- Code blocks for popular controls, ready to copy-paste
  - Example: "need to set speed between [1.0 ... 5.0] m/sec with step 0.1"

kisvegabor, 2019-03-26:
As this issue is related to examples (and not LittlevGL core) I moved to the `lv_examples` repo.

puzrin, 2019-04-16:
not stale

puzrin, 2019-05-08:
not stale

puzrin, 2019-06-15:
not stale

kisvegabor, 2019-06-15:
Reopened because will be done after v6.0 release.

kisvegabor, 2019-07-06:
Not stale

kisvegabor, 2019-07-24:
I've started to work on it [here](https://github.com/littlevgl/lv_apps/blob/master/src/lv_settings/lv_settings.c).

I've implemented the engine of the settings. 

What is working:
- Has the intended layout
- [High level definition](https://github.com/littlevgl/lv_apps/blob/e53b9fa3dd0fc217623cb1564875a0191e5d2d48/src/lv_settings/lv_settings.c#L47-L67) of menus 
- Multi level menus
- Naviagtion 
- No memory leak

What is still not required:
- Animations
- Styling
- Some special menus with slider, roller, etc

[image]

embeddedt, 2019-07-24:
@kisvegabor Looks awesome! I can't wait to try it out!

kisvegabor, 2019-07-26:
I've improved the app.
See this [example](https://github.com/littlevgl/lv_apps/tree/master/examples/lv_ex_settings)

[image]

puzrin, 2019-07-26:
> What do you think?

Some notes: it diverges from suggested design / MD guidelines in some places.

- There should be no grouped controls of different type (each on separae screen)
  - See screenshot where "motor" element slides down.
  - there should be no multiple grouped +/- items (see "time")
- List of params should be just a list of params. May be checkboxes can be embedded (not sure), but not more.

This comments are not about colors/fonts, but about layout and combinations of different items.

kisvegabor, 2019-07-26:
You put any type and number of items on a page. So you can create on one item/page if you want.

puzrin, 2019-07-26:
> You put any type and number of items on a page. So you can create one item/page if you want.

IMO such things could be done by default, to show consistent interface. Why not combine things exactly as on provided screenshots? Those are balanced and verified. In other case you risk to invent "yet another custom skin without guidelines"

embeddedt, 2019-07-26:
> you risk to invent "yet another custom skin without guidelines"

How is the number of settings on a page related to the style of the objects on a page?

puzrin, 2019-07-26:
> How is the number of settings on a page related to the style of the objects on a page?

Not number of settings, but "inlining" into list, instead of separate screen (as i posted).

kisvegabor, 2019-07-26:
@puzrin 
I'd created this example to test and show all the features. Therefore it's a little bit complex too so I name it `lv_ex_settigns_2`. 

In `lv_ex_settings_1` we can have a more simple settings menu as you suggested.

puzrin, 2019-07-26:
Oh, that's a good news :). Kick me, if you need any questions about my screenshots.

kisvegabor, 2019-07-27:
@puzrin I think there is one thing where we are thinking differently about the settings app: the styles.
@embeddedt's and my view is, it's better to let the user applying a theme because it's generic and simple. However, as the themes are generic they are not good for everything. For example, there is only one style for buttons and they work only with one font. The user can still change the theme manually or customize the settings app if he wants something different. 

A possible solution could be to have an example where the custom styles are demonstrated. What do you think?

puzrin, 2019-07-27:
> I think there is one thing where we are thinking differently about the settings app: the styles.

Since i can not be expert in everything, prefer to use existing style guides and hire designers. That reduces risks of multiple reworks.

I'd prefer more clear ratings - implementation cost (time) and result value. In this context, goal of my proposal is to get good result with minimal spends (and with guarantee to be good enough :).

puzrin, 2019-07-27:
> A possible solution could be to have an example where the custom styles are demonstrated. What do you think?

Could you clarify, what you mean as styles? If just color palette change - that's should not be problem. If change everything - this may be road without end.

embeddedt, 2019-07-27:
>  what you mean as styles? If just color palette change - that's should not be problem.

It's more than a palette change. It could also affect the transparency of objects. For example, in the "default" list the background is clearly visible but for the "material" list it's not so visible.

puzrin, 2019-07-27:
In general, it's not possible to make absolutely flexible styling, if you not reinvent something like full featured CSS engine of browser.

If we narrow case to MD only, there are 4-8 base colors (= palette), and everything else use those OR calculate result automatically (very rare or never). That's not infinite flexibility, but enough for 99% needs.

1. In real world, i would prefer to have 1 non-flexible perfect MD style, than 10 flexible "not perfect" styles :).
2. At first stage, i'd suggest to hardcode styles directly to elements, to get something working as soon as possible. Then make second iteration - rethink what can be done via themes/palettes.

embeddedt, 2019-07-27:
>  i would prefer to have 1 non-flexible perfect MD style, than 10 flexible "not perfect" styles

Doesn't that lock everyone into using MD in their applications? There would be no way to choose custom colors like there is now. I don't think that is a good decision... another great thing about LittlevGL is the amount of flexibility it gives you.

puzrin, 2019-07-27:
I do not suggest to drop "old" styles immediately. Road to MD is not fast, and it worth to make experiments in parallel with existing features.

Also i think, end users would be happy to prefer well-styled interfaces out of box, instead of mythic infinite flexibility :)

embeddedt, 2019-07-27:
> Road to MD is not fast, and it worth to make experiments in parallel with existing features.

I agree. I think we can support both.

* We can have a set of default styles that follow Material Design principles, and provide a way to change the "important" colors (i.e. primary, secondary).
   * From there we have two options. Either the documentation would explain that the primary color maps to `style.body.main_color` in the case of plain objects, or `style.text.color` in the case of text, or...
    * We provide a set of APIs for customizing the material styles (i.e. `lv_style_material_set_primary`).
* We can also allow people to manually change the properties of styles as they wish.

That should give full Material Design coherency without requiring people to dump their current styles.

kisvegabor, 2019-07-29:
> We can have a set of default styles that follow Material Design principles, and provide a way to change the "important" colors (i.e. primary, secondary).

There are only a few default styles (`lv_style_...`) and we can't use them to style every object type properly. E.g. a slider needs three very different slider specific styles: background, indicator, knob.

That's why the themes exist where every style of every object type can be adjusted. However, the themes also have limitations: only one hue value and one font can be passed. 
The [GIF posted here](https://github.com/littlevgl/lv_examples/issues/26#issuecomment-515497581) uses the material theme. IMO it looks good enough. We can surely improve it, but the result won't be so much better but it'd require a lot of effort the rework styles and themes.