lvgl/lv_demos #36: lv_test_theme_2.c exec_cb typecast to lv_bar_set_value

DeonMarais64, 2019-08-29:
https://github.com/littlevgl/lv_examples/blob/2b3c4004cb1d8060f6a04172b038131cccd89b07/lv_tests/lv_test_theme/lv_test_theme_2.c#L213

Would this not be better ?
```c
static void bar_set_value(lv_obj_t* bar, int16_t value)
{
    lv_bar_set_value(bar, value, LV_ANIM_OFF);
}
    a.exec_cb = bar_set_value;
```
As things are, each animation step could randomly spawn another animation instance ?

embeddedt, 2019-08-29:
The example was written for LittlevGL 5.3 (which didn't take a third parameter in `lv_bar_set_value`).

It would definitely cause an issue in many cases. Can you send a pull request?

DeonMarais64, 2019-08-29:
> The example was written for LittlevGL 5.3 (which didn't take a third parameter in lv_bar_set_value).
Yes, I am aware of that :)

Will do a PR, need to work through that process though. Never done that before.

kisvegabor, 2019-08-30:
@DeonMarais64 

Thank you in advance!

I summarized how to create PR [here](https://blog.littlevgl.com/2018-12-06/contributing#how-can-i-send-fixes-and-improvements).

kisvegabor, 2019-09-20:
The PR is merged.

closed 2019-09-20 by kisvegabor