lvgl/lv_demos #41: How to run hello world?

JimmyTheNerd, 2020-01-08:
I've a basic script, based on `lv_examples/lv_tutorial/1_hello_world`:

```
/* gui.c */
#include "lvgl/lvgl.h"

int main()
{	 	 
    return 0;
}

void gui(void)
{
    lv_obj_t * scr = lv_disp_get_scr_act(NULL); /* just one line from the example */
}
```

I compile it with `gcc gui.c` and get the following error:

```
/tmp/cca5tFm7.o: In function `gui':
gui.c:(.text+0x2c): undefined reference to `lv_disp_get_scr_act'
collect2: error: ld returned 1 exit status
```

Where does `lv_disp_get_scr_act()` come from and what am I missing?

Thanks in advance!

embeddedt, 2020-01-08:
You didn't compile any of the LittlevGL source files into your code.

embeddedt, 2020-01-08:
Additionally, you should be running `lv_init` and registering a display before you call any other API functions.

JimmyTheNerd, 2020-01-08:
@embeddedt I'm new to C and was expecting that all dependencies would also be compiled.

Could you please provide me the necessary lines and/or refer to the docs? I couldn't find anything in the docs, probably because it's too basic. 

Thank you!

embeddedt, 2020-01-08:
I assume that you are doing this on a PC? In that case, I strongly recommend you start with [one of the pre-built PC simulator projects](https://docs.littlevgl.com/en/html/get-started/pc-simulator.html) instead of trying to compile it by hand like this. If you don't, you will have a lot of work before you get anything functional, and it would be a waste of effort.

JimmyTheNerd, 2020-01-08:
Thanks @embeddedt. I'm using Raspbian on a Raspberry Pi 3B+. I'd like to test the look and feel under real conditions before going deeper. What do you recommend to run a demo app on that device?

embeddedt, 2020-01-08:
I recommend https://github.com/littlevgl/lv_port_linux_frame_buffer.

JimmyTheNerd, 2020-01-08:
@embeddedt Thanks! This I can compile with `make` inside the root folder and afterwards I should get `*.o`-files that are executable, correct?

embeddedt, 2020-01-08:
`.o` files are not executable; they just contain code that needs to be linked into a final executable. The final output is a file named `demo` which is executable.

JimmyTheNerd, 2020-01-08:
Thanks for the explanation, @embeddedt.

`make` worked, `./demo` as well, but I can't use the demo because the "window" seems to be an overlay which I can only click "through". The clicks have no impact. It's animation is running.

Please have a look at the screenshot. Any idea what's going on there? Thanks again!

[image]

embeddedt, 2020-01-08:
Ah... you didn't mention that you have a full X11 environment. In that case I suggest using the [SDL project](https://github.com/littlevgl/lv_sim_eclipse_sdl) instead.

The framebuffer project is designed to be used on systems without X11, where LittlevGL takes over the framebuffer and handles all input directly from the kernel.

JimmyTheNerd, 2020-01-08:
Thank you, @embeddedt. I will try the SDL project. The framebuffer will be interesting in the next step, when I run it on a headless OS. Thanks again for your help! :-)

closed 2020-01-08 by JimmyTheNerd