lvgl/lv_demos #46: Crash when calling init_all_themes()  [mbed-os 5 bare-metal]

tdjastrzebski, 2020-04-04:
I thought it would be nice to convert excellent [lv_port_stm32f769_disco_sw4stm32](https://github.com/littlevgl/lv_port_stm32f769_disco_sw4stm32) example to the latest **mbed-os 5 bare-metal**.
But as soon as `init_all_themes(0)` gets called from `lv_test_theme_2()` the system crashes with **hard fault exception**, before `lv_tick_inc()` or `lv_task_handler` is executed even once.
What could be wrong? To small stack size?

embeddedt, 2020-04-04:
Possibly. Can you use a debugger to find where it crashes?

tdjastrzebski, 2020-04-05:
I am not sure, I guess there is a lot I need to learn. ST-Util debugger I use in VS Code is not helpful here. But I managed to decode crush info using [this script](https://github.com/SenRamakri/mbed-os-example-fault-handler) I had to fix first.
```
Crash Info:
        Crash location = __exidx_start [0xFFFFB0FE] (based on PC value)
        Caller location = __exidx_start [0xFFFFFFE9] (based on LR value)
        Stack Pointer at the time of crash = [2007FE00]
        Target and Fault Info:
                Processor Arch: ARM-V7M or above
                Processor Variant: C27
                MPU or Execute Never (XN) default memory map access violation on an instruction fetch has occurred
```

embeddedt, 2020-04-05:
Maybe you should try putting a breakpoint on `main` and stepping through the code (you can step over the initialization code and step into `init_all_themes`). The crash info your script decoded seems to be incorrect.

tdjastrzebski, 2020-04-13:
@embeddedt Generally I know how to debug code. The problem was it was crashing for no apparent reason when just entering a function. I found the reason and described it [here](https://github.com/ARMmbed/mbed-os/issues/12793). Briefly, linker-reserved global vector table size was too small.
Now the execution goes into infinite loop in `Default_Handler` (infinite loop) when HAL is unlocked inside `HAL_DSI_Refresh()`
```
/* Update the display */
hdsi->Instance->WCR |= DSI_WCR_LTDCEN;
/* Process unlocked */
__HAL_UNLOCK(hdsi);
```
but at least debugger works as expected.

tdjastrzebski, 2020-04-14:
Issue solved, problem was caused by serious bug in mb-os 5: https://github.com/ARMmbed/mbed-os/issues/12793

closed 2020-04-14 by tdjastrzebski