lvgl/lv_demos #47: dev-7.0 - update demos

KammutierSpule, 2020-04-21:
Is possible to build some demo to test dev-7.0 ?

kisvegabor, 2020-04-21:
Sure, You can use `rework-7` branch.

C47D, 2020-04-27:
Can we add an option to the lv_examples configuration so the demo_widgets enable the call to  `lv_task_create(tab_changer_task_cb, 5000, LV_TASK_PRIO_LOW, NULL);`

This way we can remove the test/no-tp branch

kisvegabor, 2020-04-27:
@C47D I've added `LV_DEMO_WIDGETS_SLIDESHOW ` option. You can enable it it `lv_ex_conf.h`.

It scrolls the tabs' content vertically too.

C47D, 2020-04-27:
Thanks @kisvegabor, I fixed a typo on the `test/no-tp` branch with the commit 7c65ff1, do you think it's possible to cherry pick that commit from `test/no-tp` into `rework7`, so `test/no-tp` will be no longer necessary?

embeddedt, 2020-04-27:
@C47D I've cherry picked the change.

kisvegabor, 2020-05-19:
The new demos are merged into `master` so I close this issue.

closed 2020-05-19 by kisvegabor