lvgl/lv_demos #49: Examples and assets include paths

jozba, 2020-04-29:
Hello, 

I'm working on rework-7 branch, tried to bring-up 7.0 tests. I came across issue with lvgl.h include paths - all demos, examples and assets are using:

`#include "lvgl/lvgl.h"`

This doesn't follow style used in rest of the framework and breaks build. Shouldn't be there a construction like:

```
#if defined(LV_CONF_INCLUDE_SIMPLE)
#include "lvgl.h"
#else
#include "lvgl/lvgl.h"
#endif 
```

or is it a problem in my integration only? I use `LV_CONF_INCLUDE_SIMPLE=1` and `LV_EX_CONF_INCLUDE_SIMPLE=1` for work. 

Jozef

kisvegabor, 2020-04-29:
Hi,

You are right, it was accidentally removed.

I've just added back this functionality. Could you try it?

jozba, 2020-04-29:
Hi Gabor, 

that looks good to me. Could you also add similar fix for assets? 

Thanks, 
  Jozef

kisvegabor, 2020-05-01:
Good point. I've updated the assets too.

roddehugo, 2020-05-03:
Hi,

First, thanks for the work @kisvegabor! I've just discovered this library and it is something I've been looking for a long time.

Concerning this issue, in my current setup, I want a configuration file per application (a source file with `main.c` plus a `lv_conf.h` and a `lv_drv_conf.h` by its side). But all my applications share the same submodules, in this case, LittlevGL submodules like `lvgl`, `lv_drivers`, and `lv_examples`. 

Hence, here, shouldn't it be `LV_LVGL_H_INCLUDE_SIMPLE` instead? Like in `lv_drivers`? Otherwise, my setup does work for drivers but not for examples.

Thanks,
Hugo

roddehugo, 2020-05-04:
Actually, I have found several issues with the includes in this repository, on the master branch already. I wrote patches to apply locally. Is it something you are interested in? I can file a PR.

Here are the issues I encountered on my way:

1. the way `lvgl.h` include is handled here is not consistent with `lv_drivers`
2. so I basically applied the same schema in every include in `lv_examples`, for instance, in `lv_test_obj.h`, this patch is applied:

```diff
+#ifndef LV_EX_NO_CONF
 #ifdef LV_CONF_INCLUDE_SIMPLE
-#include "lvgl.h"
 #include "lv_ex_conf.h"
 #else
-#include "../../../lvgl/lvgl.h"
 #include "../../../lv_ex_conf.h"
 #endif
+#endif

 #if LV_USE_TESTS

+#ifdef LV_LVGL_H_INCLUDE_SIMPLE
+#include "lvgl.h"
+#else
+#include "lvgl/lvgl.h"
+#endif
+
```

3. it works fine, except for the tests that rely on the activation of features via `LV_USE_*`, like in `lv_test_cont.h`, we can find this line `#if LV_USE_CONT && LV_USE_TESTS`. But, with the above patch, `lvgl.h` comes **after** the `#if`, hence compiler complains `LV_USE_CONT` is not defined... I could have put it before, but then it would still be inconsistent with the way it is done in `lv_drivers`. So I opted for another approach, since there is main `lv_examples.h` that include `lvgl.h`. Here is the new patch for `lv_test_obj.h`, that also work for `lv_test_cont.h`:

```diff
+#include "../../../lv_examples.h"
+
+#ifndef LV_EX_NO_CONF
 #ifdef LV_CONF_INCLUDE_SIMPLE
-#include "lvgl.h"
 #include "lv_ex_conf.h"
 #else
-#include "../../../../lvgl/lvgl.h"
 #include "../../../../lv_ex_conf.h"
 #endif
+#endif

 #if LV_USE_CONT && LV_USE_TESTS
```

with `lv_examples.h` being patched with:

```diff
-#include "../lvgl/lvgl.h"
+#ifdef LV_LVGL_H_INCLUDE_SIMPLE
+#include "lvgl.h"
+#else
+#include "lvgl/lvgl.h"
+#endif
```

4. great that works as expected, but know I have not activated `LV_USE_DEMO`, and `#if` order in makes compiler complains about `LV_DEMO_WALLPAPER` not being defined, with `-Wundef` flag:

```
img_bubble_pattern.c:4:5: error: 'LV_DEMO_WALLPAPER' is not defined
```

I just switched the condition order like this:

```diff
-#if LV_DEMO_WALLPAPER && LV_USE_DEMO
+#if LV_USE_DEMO && LV_DEMO_WALLPAPER
```

and it works fine!

5. I am happy, now my setup works great, and `lv_examples` behaves like `lv_drivers`, which is a good sign of coherence in the library.

FYI: For 4., I'll file a PR; what do you think of 3.?

Have a great Monday,
Hugo

kisvegabor, 2020-05-04:
Hi Hugo,

Ups, really there are some inconsistencies here. Thank you for pointing it out.

As the examples are displayed in the [documentation](https://docs.littlevgl.com/v7/en/html/widgets/arc.html#example) too so it'd be important to keep the "header" part short.

As `master` will be replaced soon with `rework-7` let's see what are the issue there.
I pushed a fix to solve the issue you have mentioned in your first comment. 5fd053b

Does `dev-7.0` of `lvgl` works normally now with `rework-7` of `lv_examples`?

roddehugo, 2020-05-04:
> TL;DR: looks good to me although I have not tested it yet. You can consider this issue resolve from my perspective. Thanks for your support.

I am actually working with `master` here for a professional project I am involved in. It is actually the beginning of it and you will likely hear of me again :) 

I have not checked the `rework-7` branch yet, only a quick glance to not couple my code too closed to the `6.1.2` as `7.0` is coming. 

As far as I can see, with both [474dc1f0](https://github.com/littlevgl/lv_examples/commit/474dc1f0a42a74ba5629cf17e32c8a47d92377e1) and [5fd053b8](https://github.com/littlevgl/lv_examples/commit/5fd053b8d87fa988526d255cb4b42e06afc8b235), it should fix the issues I have encountered. It achieves the same result as in my [branch compared to `master`](https://github.com/littlevgl/lv_examples/compare/master...roddehugo:fix-lvgl-include).

I can live we the fix on my branch for the time being as I understand it is not worth the pain of future conflicts when merging `rework-7`.

kisvegabor, 2020-05-05:
> I am actually working with master here for a professional project I am involved in. It is actually the beginning of it and you will likely hear from me again :)

Looking forward to it! :slightly_smiling_face:

kisvegabor, 2020-05-05:
It seems all working for v7.0, so I close this issue,

closed 2020-05-05 by kisvegabor