lvgl/lv_demos #52: LV_THEME_MATERIAL_FLAG_LIGHT and LV_THEME_MATERIAL_FLAG_DARK symbols undeclared for MONO theme

C47D, 2020-05-05:
Hi,

I get the following errors while trying to compile an example application for a monochrome display.

```c
../components/lv_examples/lv_examples/src/lv_demo_widgets/lv_demo_widgets.c: In function 'color_chg_event_cb':
../components/lv_examples/lv_examples/src/lv_demo_widgets/lv_demo_widgets.c:701:25: error: 'LV_THEME_MATERIAL_FLAG_LIGHT' undeclared (first use in this function); did you mean 'LV_THEME_MATERIAL_H'?
         uint32_t flag = LV_THEME_MATERIAL_FLAG_LIGHT;
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                         LV_THEME_MATERIAL_H
../components/lv_examples/lv_examples/src/lv_demo_widgets/lv_demo_widgets.c:701:25: note: each undeclared identifier is reported only once for each function it appears in
../components/lv_examples/lv_examples/src/lv_demo_widgets/lv_demo_widgets.c:702:44: error: 'LV_THEME_MATERIAL_FLAG_DARK' undeclared (first use in this function); did you mean 'LV_THEME_MATERIAL_H'?
         if(lv_switch_get_state(sw)) flag = LV_THEME_MATERIAL_FLAG_DARK;
                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                                            LV_THEME_MATERIAL_H
```

Relevant part of my lv_config.h:
```c
#define LV_USE_THEME_MONO 1

#define LV_THEME_DEFAULT_INIT lv_theme_mono_init
#define LV_THEME_DEFAULT_COLOR_PRIMARY LV_COLOR_BLACK
#define LV_THEME_DEFAULT_COLOR_SECONDARY LV_COLOR_WHITE
#define LV_THEME_DEFAULT_FLAG 0
```

Should the example check if the application is not using the MONO theme or the value of `LV_THEME_DEFAULT_FLAG` before trying to use the `LV_THEME_MATERIAL_FLAG_DARK` and `LV_THEME_MATERIAL_FLAG_LIGHT` symbols?

kisvegabor, 2020-05-06:
Hi,
It should be fixed by  53bb179

C47D, 2020-05-06:
Thanks for the quick reply, it does compile, I haven't tested it with hardware. Solved for now.

closed 2020-05-06 by C47D