lvgl/lv_demos #56: [Request] Update README and add README to each example

C47D, 2020-05-27:
The main README still refers to the examples using LVGL v6, how can it be updated?

The purpose of a README on each example would be to list the requirements it needs (fonts, LVGL special configuration, etc.), and features it shows.

embeddedt, 2020-05-27:
I like the idea. I suggest that we just get rid of all the example-specific information in the main README and simply mention that each example has its own README.

embeddedt, 2020-05-27:
I've gone ahead and removed the outdated information in https://github.com/lvgl/lv_examples/commit/5c75a3e1b13b6739f09ec4b912204665c116e95a since it's not relevant anymore.

C47D, 2020-05-27:
We could list the examples on the main README and use markdown links to refer to each example README, something like this:

```
Examples:
lv_demo_benchmark: More info [here](relative_path_to_lv_demo_benchmark_readme).
...
```

kisvegabor, 2020-05-29:
Sounds good! I'll add it later today. I'm planning to write 1-2 sentences to each folder.  
E.g.
```
- Widget examples: Simple examples to demonstrate the features of the widgets.  See them in [lv_ex_widgets](...).
```

kisvegabor, 2020-06-02:
I've added the examples

closed 2020-06-23 by C47D