lvgl/lv_demos #58: long click the delete button cannot delete char in this keyboard screen

incity, 2020-05-29:
[image]

kisvegabor, 2020-05-31:
It's working for me like this:
[image]

embeddedt, 2020-05-31:
I was able to reproduce the issue. You have to be in numeric/symbols mode.

embeddedt, 2020-05-31:
It should be fixed in https://github.com/lvgl/lvgl/commit/af812b0c55e8e45ca129f02112c872401e87a0e6 (use the `master` branch).

incity, 2020-06-01:
> I was able to reproduce the issue. You have to be in numeric/symbols mode. I am working on a fix now.

yes

incity, 2020-06-01:
> It should be fixed in [lvgl/lvgl@af812b0](https://github.com/lvgl/lvgl/commit/af812b0c55e8e45ca129f02112c872401e87a0e6) (use the `master` branch).

thank you

closed 2020-06-02 by incity