lvgl/lv_demos #59: the lastest lv_demo_printer render effect of the four desktop items has problem.

incity, 2020-06-01:
[2020-06.zip](https://github.com/lvgl/lv_examples/files/4708992/2020-06.zip)

the new.mp4 is the lastest repo, the four desktop items' backgroud render is not smooth as the old one.

incity, 2020-06-02:
the shadow part of the desktop item has a flash when do an animation.(windows switch)

incity, 2020-06-02:
@kisvegabor could fix this?

kisvegabor, 2020-06-02:
Thanks! On the video or in my simulator I can't see any difference. I can test in on hardware tomorrow.

incity, 2020-06-02:
[image]
you can stare at the place i pointed out

incity, 2020-06-02:
[image]

incity, 2020-06-02:
seems has a background repaint?? 
[image]
the old one is like this.

kisvegabor, 2020-06-02:
Oh, got it. I tried to speed up drawing of these large images but it introduced this issue.
I'll revert it tomorrow.

kisvegabor, 2020-06-03:
I've pushed a fix into `master`.

closed 2020-06-04 by incity