lvgl/lv_demos #60: multiline editor can not move the cursor to end of line.

incity, 2020-06-01:
[image]

As shown in the figure, i can't move the cursor to the end of line use mouse click.

incity, 2020-06-01:
1. type anyone character, e.g. "a", and the cursor now is after "a", like this -> "a|"
2. click the editor, the cursor will move back, becomes  --> "|a"

kisvegabor, 2020-06-01:
@incity I've fixed it in `master`.

incity, 2020-06-02:
@kisvegabor not soloved, still have other same problems

kisvegabor, 2020-06-02:
Oh, really. There was a special case when BIDI was enabled and you clicked after the last character.
Should work now.

incity, 2020-06-02:
Hi still has other problems:
[image]
click here before popup the keyboard, and it will auto input 'o'

incity, 2020-06-02:
[image]

incity, 2020-06-04:
@kisvegabor did you see?

kisvegabor, 2020-06-04:
@incity 
The first issue (added "o" character) should be fixed here: https://github.com/lvgl/lv_examples/commit/5ac85f8faa472f7242a3591706d322219928e307

I can't reproduce the second one (can't open the keyboard again).
Can you send a video or gif?

incity, 2020-06-05:
[fb5baecb8b19b887b0b8d59cf5ad1fb4.zip](https://github.com/lvgl/lv_examples/files/4733520/fb5baecb8b19b887b0b8d59cf5ad1fb4.zip)
@kisvegabor

kisvegabor, 2020-06-05:
Fixed it, thank you!

closed 2020-06-05 by incity