lvgl/lv_demos #61: list item delimiter not show

incity, 2020-06-02:
[image]
need one line here

embeddedt, 2020-06-02:
Looks like a possible invalidation issue (the spacer above "This is a quite long text" is too long and seems to ignore the padding).

kisvegabor, 2020-06-03:
Actually it's an issue in the theme. The bottom thin line is painted to greenish when focused. But the checked button is also greenish so the line is not visible when a button is checked and focused (it's the lastly clicked).

I'm consulting with the graphics designer about how to highlight the focused button.

kisvegabor, 2020-06-04:
Should be fixed in `lvgl/master`

closed 2020-06-05 by incity