lvgl/lv_demos #74: ERROR IN lv_img_set_src from bin file

zq190kg, 2020-09-12:
hi,I use the following code,An error occurred,maybe  《lv_fs_file_t * file_p》  Parameter transfer error
void test(void)
{
	lv_obj_t *img;
	img = lv_img_create(lv_scr_act(), NULL);
	lv_img_set_src(img, "c:/1.bin");
}
[image]
[image]
[image]
but i used gif testcode,no error occurred;
void gif_demo(void)
{
lv_obj_t * img = lv_gif_create_from_file(lv_scr_act(), "c:/1.gif");
	              
}
[image]
[image]

I USE LVGL "LV_EXAMPLES"&"LVGL_NEW_FS_API"&"lv_fs_if"&"lv_lib_gif"

kisvegabor, 2020-09-14:
One thing I noticed that you should use uppercase letters as a driver.  I might mess up things internally.

zq190kg, 2020-09-14:
the same test code in lvgl master is ok,     but  in lvgl new FS API branch is error.
I found a mistake in    ”lv_img_decoder_built_in_open” (lvgl new FS API branch)
[image]

zq190kg, 2020-09-14:
in  lvgl new FS API branch ,the lv_fs_file_t->drv parameters cannot be transmitted to  fs_seek  correctly

zq190kg, 2020-09-14:
> One thing I noticed that you should use uppercase letters as a driver. I might mess up things internally.

I test uppercase letters as a driver，still error occurred in lvgl new FS API branch，but the lvgl master is ok，i think the Key points is not the uppercase,。Because I'm going to use GIF，so i must use   "LVGL_NEW_FS_API"&"lv_fs_if"&"lv_lib_gif"。

kisvegabor, 2020-09-14:
Can you test this in `lv_img_decoder.c`?
```c
        user_data->f = f;
//        _lv_memcpy_small(user_data->f, &f, sizeof(f));
```

zq190kg, 2020-09-14:
tks,user_data->f = f;  this ok
I tested “_lv_memcpy_small(user_data->f, f, sizeof(f));” before，not work

kisvegabor, 2020-09-15:
I've fixed it.

chliii, 2020-09-18:
HI,

Can you run the program now? 

I cant run the lv_lib_gif

closed 2020-10-17 by stale[bot]