lvgl/lv_demos #75: Hardfault (IntegerDivideByZero) on demo stress when using UNSCII 8 font.

C47D, 2020-09-15:
Hi,

This issue was found while working on the lv_esp32_port repo. The hardfault seems to be triggered when using symbols with the unscii 8 font set as default (the project had all the 4 font types set to unscii 8). See [here](https://github.com/lvgl/lv_port_esp32/pull/196#issuecomment-691399990) for more details.

Can symbols be used on unscii 8 font? the hardfault gets triggered with the wifi symbol.

I report this here because I think it can be sorted out on the demo code, transfer to lvgl main repo if you think it's more appropiate.

kisvegabor, 2020-09-15:
Hi,

There are no symbols in unscii8 because the FontAwesome fonts (the font used as symbols) have no monochrome version.

C47D, 2020-09-15:
I see, how can we solve it? Explicitly state that on the demo readme, exit the faulty function if the divisor is 0, check the font at runtime and display the demo without symbols?

I can make a PR with the solution if you want me to.

kisvegabor, 2020-09-15:
I've added a [fix](https://github.com/lvgl/lvgl/commit/b947465da1c6006e13b7de8fb33f692fd0376501) to return if any size is zero.

I think we can leave the demo as it is. At worst case, the symbol will be silently ignored. 
A warning is probably also added about the missing symbol.

closed 2020-10-10 by C47D