lvgl/lv_demos #78: Error width live examples

kisvegabor, 2020-10-06:
@embeddedt 
I've released v7.6.1 but now the live examples are not working. It shows 404 error instead of the examples and I can't find the html files anywhere.

BTW, if the html files are added into the gh-pages branch how can we manage more branches (other than the master)?

embeddedt, 2020-10-06:
@kisvegabor It's because it relies on the Emscripten project for building, and that project is using a version of LVGL that is "too old".

I've looked into how we can fix this. We need to push the standard release commits to the various projects with `[skip ci]` in the title. This will prevent any of the GitHub actions from running right away. Then, we need to update all the submodules on the Emscripten project. Finally, we should push another dummy commit to all the projects (with `--allow-empty`) to trigger the GitHub action.

This will ensure that the Emscripten project is up-to-date before any actions run.

embeddedt, 2020-10-06:
You can see an example of me doing this manually here: https://github.com/lvgl/lv_examples/actions

First I pushed https://github.com/lvgl/lv_examples/commit/74af0409b98765f9d83bb50048fdf0704cf140fe which was tagged to prevent the examples from being built right away. Then I went to update the Emscripten project to that commit. Finally I pushed https://github.com/lvgl/lv_examples/commit/6c08067734e0a4ee088008399bdd72af76d05813, which is empty, but still triggers GitHub actions.

kisvegabor, 2020-10-06:
Thank you for fixing it!

But if I understand correctly all examples for all lvgl version uses the same HTML files from the `gh-pages` branch. So if we will have new examples for v8 how can we separate them from v7 examples?

It seems we should have folders for at least each major release in `gh-pages`.  

We can add automatic releases to `lv_sim_emscripten` and the action update to the release branches.

What do you think?

kisvegabor, 2020-10-21:
@embeddedt 
`lv_sim_emscripten` is updated to the release branches of `lvgl`, `lv_examples`, and `lv_drivers`.

>But if I understand correctly all examples for all lvgl version uses the same HTML files from the gh-pages branch. So if we will have new examples for v8 how can we separate them from v7 examples?
>
>It seems we should have folders for at least each major release in gh-pages.

Could you comment on this?

I haven't updated the docs because wasn't sure needs to be done to make the examples work.

embeddedt, 2020-10-21:
Hmm.. somehow your previous message disappeared.

Probably it's best for Sphinx to fetch an HTML version of `lv_examples` and store them in the docs repository. I will make those changes soon.

kisvegabor, 2020-10-21:
> Probably it's best for Sphinx to fetch an HTML version of lv_examples and store them in the docs repository. I will make those changes soon.

That would be perfect!

embeddedt, 2020-10-24:
I pushed this to `latest` (not sure how to rebuild it for . An important caveat to be aware of: currently I have it set up to scan the deploy commits from https://github.com/lvgl/lv_examples/tree/gh-pages. This means that if you update the `lv_examples` submodule to a really recent commit (e.g. in the last 15 minutes), it will fail to find a deployed version (as GitHub Actions takes about 15 minutes to finish rebuilding).

I did check the release script and it doesn't appear to make any changes to submodules in `docs`, so I think releases will work fine.

-------------------------------

On another topic: I'd like to set up some type of automation so that when we push changes to the individual branches (`latest, `dev`, etc.) a built copy is automatically placed in `master` and made available on `docs.lvgl.io`. Would you be open to that? I think it would eliminate the need for a lot of these Python build scripts.

kisvegabor, 2020-10-26:
Thank you!

I needed to change `lv_conf.py` a little bit to make it work with python3.6 too. (I couldn't update sphinx to python3.7).

It compiled but the examples weren't included correctly:
[image]

If I click `index.html` only white screen appears.

The built examples in `_static` also produce only a white screen. Maybe something was missing from the `confdef.txt` of the Emscriptem project?

`lv_conf.h` goes to `8de356d` commit. 

>  Would you be open to that? I think it would eliminate the need for a lot of these Python build scripts.

Definitely yes!

embeddedt, 2020-10-26:
> It compiled but the examples weren't included correctly:

That's strange. :confused: Can you try inspecting the iframe in Developer Tools to see what its `src` attribute is?

This is what I see:

[image]

> The built examples in _static also produce only a white screen.

Hmm... they also work for me. Are you testing it using a web server or by opening the files locally?

kisvegabor, 2020-10-26:
Ahh, I was trying it locally with no webserver. I've pushed it and it works perfectly.

It's soooo great! :heart_eyes: 

So all I need to take care of when making a relase is updating docs more 15 mins after the release, right?

embeddedt, 2020-10-26:
15-20 minutes should be more than enough; in the worst case you'll just get an error while trying to rebuild the docs.

kisvegabor, 2020-10-26:
Perfect, thanks!

closed 2020-10-26 by kisvegabor

kisvegabor, 2020-10-27:
I gave up on translations because:
- can't find a really good online translation manager site
- there was no activity among translators on the forum
- it was really difficult to version control based translations. There were a lot of non-sense conflicts even only with the English "translation"

So short answer: No translation :slightly_smiling_face:

embeddedt, 2020-10-27:
Okay; for now I will write the script with the assumption that English is the only language then.

kisvegabor, 2020-10-27:
It should be fine for along time.