lvgl/lv_demos #81: Micropython examples for v7

amirgon, 2020-10-17:
Hi,

A year ago @stepansnigirev did an [excellent job](https://github.com/lvgl/lv_examples/pull/40) adding Micropython examples to lv_examples.
Since then focused moved to v7. The Micropython examples were written for v6 and are not compatible with v7.  

Looking at [LVGL documentation](https://docs.lvgl.io/latest/en/html/index.html), v6 documentation still contains all the examples from @stepansnigirev work, but v7 does not.
Unfortunately it looks like only the C examples were migrated to v7 while Micropython examples were removed.  

I think it would be a good idea to migrate Micropython examples to v7.  
The work required for migrating most examples is probably straightforward.   
Currently I don't have the time to do it myself, so I'm opening this issue just to point this up.  
In case other contributors don't have the bandwidth for this, I think this task could be tagged as "help wanted" and suggested to new contributors who are looking for a simple contribution task.

What do you think?

kisvegabor, 2020-10-20:
Thanks for opening this issue.

As we are quite close to v8 (1-2 months) maybe we can skip v7 and migrate to v8 directly a little bit later. What do you think?

amirgon, 2020-10-20:
> As we are quite close to v8 (1-2 months) maybe we can skip v7 and migrate to v8 directly a little bit later. What do you think?

I'm not sure.
- Do you have information about the adoption rate of each version?
People who currently use v7 might not want to jump to v8 since this would require many changes.
I suspect that a large number of users would keep using v7 for some time.
- It might be more difficult to migrate the examples from v6 --> v8 directly. 
- It's still not entirely clear to me how to integrate the new lv_components project into Micropython. So it might take some time before Micropython is supported in v8.

kisvegabor, 2020-10-20:
@amirgon 
It sounds reasonable. Let's go for v7. :+1:

embeddedt, 2020-10-20:
> People who currently use v7 might not want to jump to v8 since this would require many changes.
I suspect that a large number of users would keep using v7 for some time.

I am almost certain that will happen for existing projects.

amirgon, 2020-12-21:
Resolved by #85
Thank you @uraich !

closed 2020-12-21 by amirgon