lvgl/lv_demos #83: Compilation error

uraich, 2020-11-10:
When compiling the examples with the Makefile I get an error:

lv_examples/src/lv_demo_stress/lv_demo_stress.c:246:37: error: cast between incompatible function types from ‘void (*)(lv_obj_t *, uint16_t)’ {aka ‘void (*)(struct _lv_obj_t *, short unsigned int)’} to ‘void (*)(void *, lv_anim_value_t)’ {aka ‘void (*)(void *, short int)’} [-Werror=cast-function-type]
  246 |             lv_anim_set_exec_cb(&a, (lv_anim_exec_xcb_t)lv_arc_set_end_angle);
      |                                     ^
lv_examples/src/lv_demo_stress/lv_demo_stress.c:256:37: error: cast between incompatible function types from ‘void (*)(lv_obj_t *, uint16_t)’ {aka ‘void (*)(struct _lv_obj_t *, short unsigned int)’} to ‘void (*)(void *, lv_anim_value_t)’ {aka ‘void (*)(void *, short int)’} [-Werror=cast-function-type]
  256 |             lv_anim_set_exec_cb(&a, (lv_anim_exec_xcb_t)lv_linemeter_set_angle_offset);

I got rid of it adding -Wno-cast-function-type
I am using gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0

uraich, 2020-11-10:
When I run lv_demo_stress, I get a segmentation violation:
eading symbols from demo...
(gdb) r
Starting program: /opt/ucc/micros/esp32/lvgl/simulator/demo 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffefe73700 (LWP 49652)]
[New Thread 0x7fffef672700 (LWP 49653)]
[New Thread 0x7fffeee71700 (LWP 49654)]
[New Thread 0x7fffee670700 (LWP 49655)]
[New Thread 0x7fffedd1a700 (LWP 49657)]
User: mem leak since start: 0, frag:   1 % 	(lv_demo_stress.c #79 obj_test_task_cb())

Thread 1 "demo" received signal SIGSEGV, Segmentation fault.
_vsnprintf (buffer=buffer@entry=0x5555558eedf8 <work_mem_int+11672> "Formatted:\n100 \377\377\377", 
    maxlen=maxlen@entry=19, format=<optimized out>, format@entry=0x5555556dd40e "Formatted:\n%d %s", 
    va=va@entry=0x7fffffffd630, out=0x5555555be0c0 <_out_buffer>)
    at /opt/ucc/micros/esp32/lvgl/simulator/lvgl/src/lv_misc/lv_printf.c:814
814	                    unsigned int l = _strnlen_s(p, precision ? precision : (size_t) -1);
(gdb) 

When I compile the program with cmake, I get no compilation error but the program seg-faults the same way

kisvegabor, 2020-11-10:
Do you use the latest version of lvgl and lv_examples?

It complies and works well for me.

uraich, 2020-11-10:
Hmmm, I downloaded lv_sim_eclipse_sdl:
git clone https://github.com/lvgl/lv_sim_eclipse_sdl.git --recursive
then I tried to compile with "make"
This produces the compilation error on my Ubuntu 20.04 system. When creating the Makefile with cmake I don't have the problem. 
Then I replaced lv_demo_widgets() by  lv_demo_stress() and re-compiled.
This works ok now. It seems I did not have the very latest lv_examples and/or lvgl
Another question: Do you have an example how to use lv_task_create in MicroPython? I get seg-faults when setting values from within the task:
https://github.com/uraich/lv_mpy_examples/blob/main/widgets/arc/ex2_arc.py
You may see immediately what I am doing wrong?

kisvegabor, 2020-11-12:
I've just tried a fresh clone and still works. Anyway, the warnings seem correct because really there is a "function type cast"  but I don't how to do it correctly then. 

> Another question: Do you have an example how to use lv_task_create in MicroPython? I get seg-faults when setting values from within the task:

@amirgon could you help with this?

uraich, 2020-11-12:
It seems that the warning has been added with gcc version 8. If you run an older version you will not get the warning. As far as I understand it is the first parameter, which is an lv_obj_type * in case of lv_arc_set_end_angle but a void * in case of lv_anim_exec_xcb_t which causes the warning.
Unfortunately I also don't know how to get rid if the problem except of disabling the warning.

amirgon, 2020-11-12:
> > Another question: Do you have an example how to use lv_task_create in MicroPython? I get seg-faults when setting values from within the task:
> 
> @amirgon could you help with this?

@uraich The point is that **you are not supposed to use the `user_data`**, [as explained in the README](https://github.com/lvgl/lv_binding_micropython#user-content-callbacks:~:text=The%20user_data%20field%20must%20not%20be,to%20hold%20pointers%20to%20Micropython%20objects.).  
`user_data` is used internally by the Micropython bindings to keep a reference to a Python callable object and invoke it in the callback.
You don't need `user_data` for passing data into a  Micropython callback. There are multiple ways you can pass data into a callback. `lambda` is one example. Another example is calling an object method, and using other object attributes to pass data to the callback.

So you can rewrite your code like this:

```python
def arc_loader(task, arc):    
    angle = arc.get_value()
    print("angle: ",angle)
    angle += 5
    if angle >= 100:
        task._del()
    else:
        arc.set_value(angle)
    
# create an arc which acts as a loader

arc = lv.arc(lv.scr_act(),None)
arc.set_bg_angles(0,360)
arc.set_rotation(270)
arc.align(None,lv.ALIGN.CENTER,0,0)

arc.set_value(10)
lv.task_create(lambda task: arc_loader(task, arc), 20, lv.TASK_PRIO.LOWEST, {})
```
In Micropython it is recommended to use `task_create_basic` like this:

```python
task = lv.task_create_basic()
task.set_cb(lambda task: arc_loader(task, arc))
task.set_period(20)
task.set_prio(lv.TASK_PRIO.LOWEST)
```

While both are `task_create` and `task_create_basic` are supported today, the `task_create_basic` hides the `user_data`.  
In  `task_create` you need to pass it as an empty dict for the binding internal usage.

uraich, 2020-11-13:
Hi Amir,
Thanks for your help. I must admit that I had some difficulty to understand this part of the README. An example is therefore of great help. I slightly adapted your code to my needs and things work perfectly fine.

uraich, 2020-11-14:
Hi Amir,
I read the README very carefully and still have some problems to understand how to use callbacks:
First some typos:
The callback convetion assumes the following: ("n" missing)
Then you should be consistent:
lvgl.anim_set_custom_exec_cb(anim, lambda anim, val, obj=obj: obj.set_y(val))
lv.anim_set_exec_cb(anim, obj, obj.set_y)
Either lvgl... or lv.... in both cases.
From the user perspective, any python callable object (such as python regular function, class function, lambda etc.) can be user as an lvgl callbacks. (should read callback without s)

Then my programming problem:
When calling 
# create an animation
a=lv.anim_t()
a.init()
print("set custom exec ")
lv.anim_set_exec_cb(a, arc, arc.set_value)
I get AttributeError: 'module' object has no attribute 'anim_set_exec_cb'

When calling:
a.set_exec_cb(a,arc,arc.set_value)
I get TypeError: function takes 2 positional arguments but 4 were given
How  do I do this correctly?
Thanks Uli

amirgon, 2020-11-14:
> First some typos:
> The callback convetion assumes the following: ("n" missing)
> Then you should be consistent:
> lvgl.anim_set_custom_exec_cb(anim, lambda anim, val, obj=obj: obj.set_y(val)) [...]

Please send me a pull request with these fixes.
English is not my native language so I appreciate any corrections.

> When calling:
> a.set_exec_cb(a,arc,arc.set_value)
> I get TypeError: function takes 2 positional arguments but 4 were given

You should use 

```python
a.set_exec_cb(lv.arc.set_value)
```

- You don't need to pass `a` again to `set_exec_cb`.
- You don't need to pass `arc` here. You should call `a.set_var(arc)` 
- You need to pass `lv.arc.set_value` (and not `arc.set_value`) because you are passing a function pointer to `set_value`.

Here is a complete example:

```python
arc = lv.arc(scr)
a=lv.anim_t()
a.init()
a.set_exec_cb(lv.arc.set_value)
a.set_var(arc)
a.set_time(1000)
a.set_values(0, 100)
lv.anim_t.start(a)
```

The reason we call `lv.anim_t.start(a)` and not `a.start()` is because of ambiguity.  
The struct `lv_anim_t` [already contains an integer member called `start`](https://github.com/lvgl/lvgl/blob/6060d4d1f6bf3530ae9f4b14d72ce13e3b8b9652/src/lv_misc/lv_anim.h#L84), which shadows the member function `lv_anim_start`.

uraich, 2020-11-15:
I did a copy / paste of the code above and tried it. 
I get:
SyntaxError: Cannot convert 'function' to pointer!
when trying to call 
a.set_exec_cb(lv.arc.set_value)

amirgon, 2020-11-15:
> SyntaxError: Cannot convert 'function' to pointer!
> when trying to call
> a.set_exec_cb(lv.arc.set_value)

You are right. In recent versions member functions in all forms can no longer be converted to pointers.
I would need to look into it.

To overcome this, you can use `set_custom_exec_cb` instead and delegate the call to `arc.set_value`.
For example:

```python
arc = lv.arc(scr)
a=lv.anim_t()
a.init()
a.set_custom_exec_cb(lambda a, val: arc.set_value(val))
a.set_time(1000)
a.set_values(0, 100)
lv.anim_t.start(a)
```

Do not call `set_var` when using `set_custom_exec_cb`.

closed 2020-12-19 by stale[bot]