lvgl/lv_demos #84: Workflow issue

kisvegabor, 2020-11-17:
On the last release this error happened:
https://github.com/lvgl/lv_examples/actions/runs/367900030

I've fixed it here but obviously, it's not the best way. :slightly_frowning_face: 
https://github.com/lvgl/lv_examples/commit/789422b51d4fb33154f7b4dfafde148a6eff23a6

embeddedt, 2020-11-17:
I wonder why I wasn't notified about the build failure on this repository. Maybe because I didn't create it? :confused:

It's because some of our actions are out of date; I updated them to the latest versions on the Emscripten repository and that seemed to solve it, so hopefully the same fix works here.

embeddedt, 2020-11-17:
Looks like it built successfully with my fix.

closed 2020-11-17 by embeddedt

kisvegabor, 2020-11-18:
Awesome, thank you!