lvgl/lv_demos #87: Deploy error with music player

kisvegabor, 2020-12-02:
I've just merged the music player demo to master and got this error from the depoly script:
```
main.c:76:5: error: implicit declaration of function 'lv_demo_music_list_btn_check' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
    CHOSEN_DEMO();
    ^
<command line>:10:21: note: expanded from here
#define CHOSEN_DEMO lv_demo_music_list_btn_check
```

Seemingly it interprets the image files as demo.

@embeddedt could you take a look?

embeddedt, 2020-12-02:
It searches for any global function in `lv_examples` that starts with `lv_` and returns `void`.

https://github.com/lvgl/lv_examples/blob/1dd94b8989583c35d430acdd532fa68592d7bce9/src/lv_demo_music/lv_demo_music_list.h#L31

Either this function would need to be changed to `static` or I would have to come up with a better heuristic (maybe checking for no parameters?).

kisvegabor, 2020-12-02:
I can add `_lv` prefix to these functions. (only tomorow)

embeddedt, 2020-12-03:
I've fixed this now; there are a few other unrelated errors caused by the Emscripten project being slightly out of date. The examples should deploy soon.

closed 2020-12-03 by embeddedt

kisvegabor, 2020-12-04:
Great! Thank you.