lvgl/lv_demos #90: lvgl.h path

C47D, 2020-12-20:
There are some includes like this `#include "lvgl/lvgl.h"`, should we update them to:

```c
#ifdef LV_CONF_INCLUDE_SIMPLE
#include "lvgl.h"
#else
#include "lvgl/lvgl.h"
#endif
```

C47D, 2020-12-21:
Forgot to work on my fork of lv_examples, here's the diff:

```diff
diff --git a/lv_examples.h b/lv_examples.h
index 7195d36..8d04bbf 100644
--- a/lv_examples.h
+++ b/lv_examples.h
@@ -13,7 +13,7 @@ extern "C" {
 /*********************
  *      INCLUDES
  *********************/
-#if defined(LV_LVGL_H_INCLUDE_SIMPLE)
+#if defined(LV_CONF_INCLUDE_SIMPLE)
 #include "lvgl.h"
 #else
 #include "../lvgl/lvgl.h"
diff --git a/src/lv_demo_music/assets/img_lv_demo_music_btn_play.c b/src/lv_demo_music/assets/img_lv_demo_music_btn_play.c
index b8685b5..88dbeaa 100644
--- a/src/lv_demo_music/assets/img_lv_demo_music_btn_play.c
+++ b/src/lv_demo_music/assets/img_lv_demo_music_btn_play.c
@@ -1,4 +1,9 @@
+#ifdef LV_CONF_INCLUDE_SIMPLE
+#include "lvgl.h"
+#else
 #include "lvgl/lvgl.h"
+#endif
+
 #include "../lv_demo_music.h"
 #if LV_USE_DEMO_MUSIC
```

kisvegabor, 2020-12-21:
`LV_LVGL_H_INCLUDE_SIMPLE` should be correct here. 
`LV_CONF_INCLUDE_SIMPLE` is used only in [lv_conf_internal.h](https://github.com/lvgl/lvgl/blob/master/src/lv_conf_internal.h#L43).

C47D, 2020-12-21:
I see, where's `LV_LVGL_H_INCLUDE_SIMPLE` defined?

kisvegabor, 2020-12-21:
> I see, where's LV_LVGL_H_INCLUDE_SIMPLE defined?

It should be defined in the project's config file too similarly to `LV_CONF_INCLUDE_SIMPLE`.

C47D, 2020-12-21:
Ok, do you want me to send a PR to fix that in here?

kisvegabor, 2020-12-22:
I've just fixed it in `img_lv_demo_music_btn_play.c`.

C47D, 2020-12-22:
Thanks, the lv_port_esp32 repo now builds!

closed 2020-12-22 by C47D