lvgl/lv_demos #91: Unable to allocate memory when performing benchmark

embeddedpenguin, 2020-12-22:
Here is my trace:

```
[TRACE] [../thirdparty/lvgl/src/lv_core/lv_obj.c:177:lv_init] -- lv_init started
[INFO]  [../thirdparty/lvgl/src/lv_core/lv_obj.c:236:lv_init] -- lv_init ready
[TRACE] [../thirdparty/lvgl/src/lv_core/lv_obj.c:278:lv_obj_create] -- Screen create started
[INFO]  [../thirdparty/lvgl/src/lv_core/lv_obj.c:469:lv_obj_create] -- Object create ready
[TRACE] [../thirdparty/lvgl/src/lv_core/lv_obj.c:278:lv_obj_create] -- Screen create started
[INFO]  [../thirdparty/lvgl/src/lv_core/lv_obj.c:469:lv_obj_create] -- Object create ready
[TRACE] [../thirdparty/lvgl/src/lv_core/lv_obj.c:278:lv_obj_create] -- Screen create started
[INFO]  [../thirdparty/lvgl/src/lv_core/lv_obj.c:469:lv_obj_create] -- Object create ready
[TRACE] [../thirdparty/lvgl/src/lv_widgets/lv_label.c:78:lv_label_create] -- label create started
[TRACE] [../thirdparty/lvgl/src/lv_core/lv_obj.c:310:lv_obj_create] -- Object create started
[INFO]  [../thirdparty/lvgl/src/lv_core/lv_obj.c:469:lv_obj_create] -- Object create ready
[INFO]  [../thirdparty/lvgl/src/lv_widgets/lv_label.c:166:lv_label_create] -- label created
[TRACE] [../thirdparty/lvgl/src/lv_widgets/lv_label.c:78:lv_label_create] -- label create started
[TRACE] [../thirdparty/lvgl/src/lv_core/lv_obj.c:310:lv_obj_create] -- Object create started
[INFO]  [../thirdparty/lvgl/src/lv_core/lv_obj.c:469:lv_obj_create] -- Object create ready
[INFO]  [../thirdparty/lvgl/src/lv_widgets/lv_label.c:166:lv_label_create] -- label created
[TRACE] [../thirdparty/lvgl/src/lv_core/lv_obj.c:310:lv_obj_create] -- Object create started
[INFO]  [../thirdparty/lvgl/src/lv_core/lv_obj.c:469:lv_obj_create] -- Object create ready
[TRACE] [../thirdparty/lvgl/src/lv_core/lv_obj.c:310:lv_obj_create] -- Object create started
[INFO]  [../thirdparty/lvgl/src/lv_core/lv_obj.c:469:lv_obj_create] -- Object create ready
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_anim.c:94:lv_anim_start] -- animation create started
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_anim.c:121:lv_anim_start] -- animation created
[TRACE] [../thirdparty/lvgl/src/lv_core/lv_obj.c:310:lv_obj_create] -- Object create started
[INFO]  [../thirdparty/lvgl/src/lv_core/lv_obj.c:469:lv_obj_create] -- Object create ready
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_anim.c:94:lv_anim_start] -- animation create started
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_anim.c:121:lv_anim_start] -- animation created
[TRACE] [../thirdparty/lvgl/src/lv_core/lv_obj.c:310:lv_obj_create] -- Object create started
[INFO]  [../thirdparty/lvgl/src/lv_core/lv_obj.c:469:lv_obj_create] -- Object create ready
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_anim.c:94:lv_anim_start] -- animation create started
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_anim.c:121:lv_anim_start] -- animation created
[TRACE] [../thirdparty/lvgl/src/lv_core/lv_obj.c:310:lv_obj_create] -- Object create started
[INFO]  [../thirdparty/lvgl/src/lv_core/lv_obj.c:469:lv_obj_create] -- Object create ready
[WARN]  [../thirdparty/lvgl/src/lv_misc/lv_mem.c:208:lv_mem_alloc] -- Couldn't allocate memory
[WARN]  [../thirdparty/lvgl/src/lv_misc/lv_mem.c:323:lv_mem_realloc] -- Couldn't allocate memory

```

I'm well under the memory limit of my MCU. I'm using a same54n19a which has 512KB of flash and 192KB of ram. Here is my memory usage:
```
   text	   data	    bss	    dec	    hex	filename
 242140	   2232	  64760	 309132	  4b78c	rev_0.elf
```

Any ideas why the benchmark won't run?

kisvegabor, 2020-12-22:
What is `LV_MEM_SIZE` in `lv_conf.h`?

embeddedpenguin, 2020-12-22:
I have LV_MEM_SIZE defined as `(4U * 1024U)`

embeddedpenguin, 2020-12-22:
Oh I doubled that and it started working. Sorry... =( 

Do I need touch support for this benchmark to work?

embeddedt, 2020-12-22:
No; it should start without any touch. If it's not doing that something else is wrong.

embeddedpenguin, 2020-12-22:
Hmmm. It seems it endlessly gets stuck in a loop of starting the task handler.

```
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:74:lv_task_handler] -- lv_task_handler started
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:192:lv_task_handler] -- lv_task_handler ready
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:74:lv_task_handler] -- lv_task_handler started
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:192:lv_task_handler] -- lv_task_handler ready
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:74:lv_task_handler] -- lv_task_handler started
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:192:lv_task_handler] -- lv_task_handler ready
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:74:lv_task_handler] -- lv_task_handler started
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:192:lv_task_handler] -- lv_task_handler ready
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:74:lv_task_handler] -- lv_task_handler started
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:192:lv_task_handler] -- lv_task_handler ready
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:74:lv_task_handler] -- lv_task_handler started
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:192:lv_task_handler] -- lv_task_handler ready
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:74:lv_task_handler] -- lv_task_handler started
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:192:lv_task_handler] -- lv_task_handler ready
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:74:lv_task_handler] -- lv_task_handler started
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:192:lv_task_handler] -- lv_task_handler ready
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:74:lv_task_handler] -- lv_task_handler started
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:192:lv_task_handler] -- lv_task_handler ready
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:74:lv_task_handler] -- lv_task_handler started
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:192:lv_task_handler] -- lv_task_handler ready
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:74:lv_task_handler] -- lv_task_handler started
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:192:lv_task_handler] -- lv_task_handler ready
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:74:lv_task_handler] -- lv_task_handler started
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:192:lv_task_handler] -- lv_task_handler ready
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:74:lv_task_handler] -- lv_task_handler started
[TRACE] [../thirdparty/lvgl/src/lv_misc/lv_task.c:192:lv_task_handler] -- lv_task_handler ready
```

embeddedt, 2020-12-22:
The current message is misleading - "ready" is at the end of the task handler. This is probably just a translation issue. 

https://github.com/lvgl/lvgl/blob/07b6d93de4e6afaa743f8146812bccae2d4041c5/src/lv_misc/lv_task.c#L192

Are you calling `lv_tick_inc` anywhere?

embeddedpenguin, 2020-12-22:
It seems that the problem was that I am a total and complete idiot! I tried changing to a custom tick function and it wasn't working, so I switched back to lv_tick_inc(1) and it worked! Thanks!

closed 2020-12-22 by embeddedpenguin