lvgl/lv_demos #92: Declaring 4 times bigger Canvas Buffer

Player-1-TR, 2021-01-06:
Am I missing something I'm not sure.
When I look up the canvas examples Canvas's buffer is always 4 times bigger than the actual pixel buffer.
For example 
[https://github.com/lvgl/lv_examples/blob/master/src/lv_ex_widgets/lv_ex_canvas/lv_ex_canvas_1.c](url)
src/lv_ex_widgets/lv_ex_canvas/lv_ex_canvas_2.c
line 16:
`static lv_color_t cbuf[LV_CANVAS_BUF_SIZE_TRUE_COLOR(CANVAS_WIDTH, CANVAS_HEIGHT)];`

For a RGBA8888 format:
`LV_CANVAS_BUF_SIZE_TRUE_COLOR(CANVAS_WIDTH, CANVAS_HEIGHT)`
means this;
`LV_IMG_BUF_SIZE_TRUE_COLOR(w, h) ((LV_COLOR_SIZE / 8) * w * h)`
then w=200 h=150;
`((32 / 8) * w * h) = 120 000 array elements`
lv_color_t 4 bytes;
`4 * 120 000 = 480 000 bytes!`

Raw data should be
`RGBA8888 x 150 x 200 = 120 000 bytes?`

Is it a issue or not?

embeddedt, 2021-01-06:
I'm pretty sure you're correct; I think this buffer should be declared with `uint8_t` not `lv_color_t`.

kisvegabor, 2021-01-09:
That correct, it really gives the result in bytes.
I've just fixed the example.

Thank you for the report.

commit 318be44c (2021-01-09): Fix canvas size

Fixes #92
closed 2021-01-09 by kisvegabor