lvgl/lv_demos #96: gauge example offset

lutzbickhardt, 2021-02-12:
the origin of the three hands is not in the same when x/y is other than 0.
the scale is not centered wrt to background circle.
[image]

embeddedt, 2021-02-12:
Hmm.. which example is this and what screen size are you using?

lutzbickhardt, 2021-02-12:
lv_ex_gauge with     lv_gauge_set_needle_img     (gauge, &img_hand, 20, 20);
res 480x480 on an iMac, same problem but different origins on a TTGO and raspi
[image]

[image]
(emailed to gabor as well, sorry only
 just realized how to report an issue on guthub)

Lutz

lutzbickhardt, 2021-02-12:
the blue arrow is set per random number, magenta and yellow are fixed

kisvegabor, 2021-02-15:
Hi Lutz,

> (emailed to gabor as well, sorry only just realized how to report an issue on guthub)

No problem :slightly_smiling_face: 

I've just tested it on my PC and it works correctly here. However, if you use the `img_hand` image you shouldn't change the `4,4` to `20,20` because it's the pivot point of the image and the image will be rotated around this point.

It doesn't explain why the whole scale is shifted on its background. Please send the whole code snippet to reproduce the issue.

lutzbickhardt, 2021-02-16:
[image]

Hi Gabor,

this is the gauge with lv_gauge_set_needle_img     (gauge, &img_hand, 4, 4); as in your example.

Still, the scale is shifted and since  lv_gauge_set_needle_img is not set PER hand, I would assume them to be set to the SAME origin.

the code is as follows:

void lv_ap_gauge_3(lv_obj_t *& gauge) {
    printf("lv_ap_gauge_3\n");
  
    //Describe the color for the needles
    static lv_color_t needle_colors[3];
    needle_colors[0] = LV_COLOR_BLUE;
    needle_colors[1] = LV_COLOR_ORANGE;
    needle_colors[2] = LV_COLOR_PURPLE;

    LV_IMG_DECLARE(img_hand);

    // Using the scale style properties
    static lv_style_t style;
    lv_style_init(&style);

    //Set a background color and a radius
    //lv_style_set_radius(&style, LV_STATE_DEFAULT, 5);
    
    //lv_style_set_bg_opa(&style, LV_STATE_DEFAULT, LV_OPA_COVER);  //leaves frgments on redraw???
    lv_style_set_bg_color               (&style, LV_STATE_DEFAULT, LV_COLOR_SILVER);
    
    //Set some paddings
    lv_style_set_pad_inner              (&style, LV_STATE_DEFAULT, 20);
    lv_style_set_pad_top                (&style, LV_STATE_DEFAULT, 20);
    lv_style_set_pad_left               (&style, LV_STATE_DEFAULT, 5);
    lv_style_set_pad_right              (&style, LV_STATE_DEFAULT, 5);
    
    lv_style_set_line_color             (&style, LV_STATE_DEFAULT, LV_COLOR_WHITE);
    lv_style_set_line_width             (&style, LV_STATE_DEFAULT, 2);

    lv_style_set_scale_grad_color       (&style, LV_STATE_DEFAULT, LV_COLOR_RED);
    lv_style_set_scale_end_line_width   (&style, LV_STATE_DEFAULT, 4);
    lv_style_set_scale_end_border_width (&style, LV_STATE_DEFAULT, 4);
    lv_style_set_scale_end_color        (&style, LV_STATE_DEFAULT, LV_COLOR_RED);

    //lv_style_set_scale_end_line_color(&style, LV_STATE_DEFAULT, LV_COLOR_RED); //does not exist ?!?
  
    //Create a gauge
    //lv_obj_t * gauge = lv_gauge_create(lv_scr_act(), NULL);
    gauge = lv_gauge_create     (lv_scr_act(), NULL);
    lv_obj_add_style            (gauge, LV_GAUGE_PART_MAIN, &style);

    lv_gauge_set_scale          (gauge, 300, 21, 6); //angle, line_num, label_cnt);
    
    lv_gauge_set_needle_count   (gauge, 3, needle_colors);
    lv_gauge_set_needle_img     (gauge, &img_hand, 4, 4);

    lv_obj_set_size             (gauge, LV_VER_RES, LV_HOR_RES);
    lv_obj_align                (gauge, NULL, LV_ALIGN_CENTER, 0, 0);

    //Allow recoloring of the images according to the needles' color
    lv_obj_set_style_local_image_recolor_opa(gauge, LV_GAUGE_PART_NEEDLE, LV_STATE_DEFAULT, LV_OPA_COVER);

    lv_obj_set_event_cb         (gauge, generic_event_cb);   /*Assign an event callback*/

    //defaults for hand
    lv_gauge_set_value(gauge, 0, 45);
    lv_gauge_set_value(gauge, 1,  0);
    lv_gauge_set_value(gauge, 2, 80);
}

lutzbickhardt, 2021-02-16:
as you see, I have used gauge as a parameter to the call with
            lv_obj_t * gauge;
            lv_gauge_set_value(gauge, 0, xx);

kisvegabor, 2021-02-17:
Thanks for the code. I could reproduce it and pushed a fix.

lutzbickhardt, 2021-02-17:
ok, that solved part of the problem; however:
- the scale is still off,
- with lv_gauge_set_needle_img (..., 4,4) all three hands are in the middle; with 24, 4 see below ->,
- all three hands should originate at the same position, no matter whether (..., 4,4) or (..., 24, 4),
- (it might be a good idea to substitute lv_gauge_set_needle_img (gauge, &img_hand, ...) by
   lv_gauge_set_needle_img (gauge, needle_1/2/3, &img_hand, ...);
- the artefact is only created when the blue arrow moves from a position between the yellow and the full blue hand, to its actual position above the blue hand (the artefact is the full hand not the only partially drawn one above it).

[image]

kisvegabor, 2021-02-19:
> the scale is still off,

It's because you set different padding on the top.

> with lv_gauge_set_needle_img (..., 4,4) all three hands are in the middle; with (..., 24, 4) see below ->,

They should remain in the middle. What you can control is the point on which the needles are rotated. With (100, 4) the needle will be rotated around its ~half width. See here how the yellow image is shifted "inside"

[image]

> (it might be a good idea to substitute lv_gauge_set_needle_img (gauge, &img_hand, ...) by
lv_gauge_set_needle_img (gauge, needle_1/2/3, &img_hand, ...);

I agree. In v8 it will work like this!

> the artefact is only created when the blue arrow moves from a position between the yellow and the full blue hand, to its actual position above the blue hand (the artefact is the full hand not the only partially drawn one above it).

I've pushed another fix to `master`. Is it working well now?

lutzbickhardt, 2021-02-19:
Hi Gabor,

sorry, I completely misunderstood the meaning of lv_gauge_set_needle_img; as you say, x/y specifies the center of rotation within the hand, I thought is was the offset of the needles end from the center of the gauge.
And I had interpreted the margin to be outside of the backgound circle (but missed to realize they were not).

The artefacts are gone!

That leaves one open point: how do I make the needle longer?
The question arises because I use 240x240 res on the TTGO and 480x480 on the Mac. And because I cannot find LV_ZOOM any more (I believe I have seen it somewhere) I somehow have to enlarge the needle. Maybe a zoom factor for any image generated by the c-pgm would help?

lutzbickhardt, 2021-02-19:
sorry, I forgot: the markers at 80 and 100 are still green! shouldt they be red as in your v8 example?

kisvegabor, 2021-02-22:
To make the image longer, the easiest is to use a larger image. :slightly_smiling_face: 

> sorry, I forgot: the markers at 80 and 100 are still green! shouldt they be red as in your v8 example?
Try this:
```c
  static lv_style_t style2;
  lv_style_init(&style2);
  lv_style_set_scale_end_color(&style2, LV_STATE_DEFAULT, LV_COLOR_RED);
  lv_obj_add_style            (gauge, LV_GAUGE_PART_MAJOR, &style2);
```

In v8 it will be much simpler.

lutzbickhardt, 2021-02-23:
ok, it worked! 
So I consider this case closed!
Looking forward to v8!

closed 2021-02-23 by lutzbickhardt