lvgl/lv_demos #102: lv_demo_music_main.c looks wrong

photomoose, 2021-03-31:
It looks like `lv_demo_music_main.c` has been commented out in its entirety and any occurrence of the word `signal` has been replaced with the string "Call the ancestor's event handler".

Was this intentional?

embeddedt, 2021-04-01:
I believe some of the v7 demos are still being updated, so the WIP version is commented out to prevent the build from breaking. At least, this is what happened for `lv_demo_widgets`.

kisvegabor, 2021-04-01:
`lv_demo_music` is still really not updated. I also saw "Call the ancestor's event handler", it probably comes from a Search/Replace issue.

I will update it soon.

embeddedt, 2021-05-26:
It appears that `lv_demo_music` compiles now, but it seems to have grown in flash size since v7.

@kisvegabor Maybe it's worth adding a setting that reduces the number of images being used? My binary is over 800KB which means this is just barely going to fit into the 1MB of flash on STM32F7.

kisvegabor, 2021-05-28:
I've pushed some changes to the music player demo. 
The number of fonts is reduced and nothing else significant hasn't changed.

Can you run `nm` to see what causes the increase in size?

embeddedt, 2021-05-30:
```
080cd8b8 00003660 R img_lv_demo_music_btn_list_pause_map
080a53f8 00003660 R img_lv_demo_music_btn_list_play_map
080a8a58 00003c10 R img_lv_demo_music_btn_next_map
080e29a4 00003c10 R img_lv_demo_music_btn_prev_map
0809ebec 00005f0c R img_lv_demo_music_btn_pause_map
080dc198 00005f0c R img_lv_demo_music_btn_play_map
080958e8 00009304 R img_lv_demo_music_logo_map
08104c04 0000b280 R img_lv_demo_music_wave_bottom_map
080d0f18 0000b280 R img_lv_demo_music_wave_top_map
20023268 00010000 b work_mem_int.5420
080e65b4 0001e140 R img_lv_demo_music_cover_1_map
080af778 0001e140 R img_lv_demo_music_cover_2_map
08075948 0001e140 R img_lv_demo_music_cover_3_map
```

Looks like images are the main culprit.

kisvegabor, 2021-05-30:
But images weren't changed in v8. Form the image sizes it seems you set 32 bit color depth. It ~doubles the image sizes.

embeddedt, 2021-05-30:
Ah, yes, I changed that for another project and forgot to switch it back. That explains it. Thanks! :slightly_smiling_face:

kisvegabor, 2021-05-31:
As the original question and @embeddedt's question should be solved I close this issue.

closed 2021-05-31 by kisvegabor