lvgl/lv_demos #105: MAX resolution display on 9486

pawelmielnik, 2021-04-19:
Hi, 
I just started to use LVGL. The first program I run on 9341, TFT and Touchscreen works ok. The resulution of tft and touch is 320x240.
I tried to run a program on 9486 which is 480x320.  I connected 9486 : touch recognise full 480x320 matrix, unfortunately screen works in 320x240 mode. Of coures I changed in menu config/component/lvgl configutation  resulution to 480x320 but no result. 
Kconfig keeps 480x 320  resolution. Is there any other place I have to enter the resolution?

pawelmielnik, 2021-04-19:
Hi,
What I also noticed, when I call again  lv_demo_widgets() instead of my short program, 9486 runs in 480x320 what is good. So 
where is a issue?

kisvegabor, 2021-04-19:
Are you using the lv_port_esp32 repository?

pawelmielnik, 2021-04-19:
Yes, I'm using lv_port_esp32.  I'm not experienced in this whole solution, in the main i swaped lv_demo_widgets with my program. 
Maybe this is not the right  approach and  gettig rid of  lv_demo_widgets destroyed  also some dependencies elsewhere?

pawelmielnik, 2021-04-19:
Gabor, I have found soimething. In file lv_example\src\lv_demo_widgets.py i have found lot of references to resolution. Is it the file i missed?  If I'm correct , how can I develope similar file to my project?

ataweg, 2021-04-19:
Take a look at https://github.com/ataweg/lvgl/blob/master/examples/arduino/LVGL_Demo_Widget/LVGL_Demo_Widget.ino

pawelmielnik, 2021-04-20:
Hi, Thanks for help. I started the new project from beginnig , I have no idea what I had done previously wrong, but now application recognise 480x320.

pawelmielnik, 2021-04-20:
Resolved

closed 2021-04-20 by pawelmielnik