lvgl/lv_demos #116: Using LV_ demo_ benchmark(); During template compilation, the following contents will appear, as if img is not defined?

superCong, 2021-08-05:
Using LV_ demo_ benchmark(); During template compilation, the following contents will appear, as if img is not defined?

ESP32:
`Building in release mode
Compiling .pio/build/esp32dev/libf95/lv_examples/lv_demo_benchmark/lv_demo_benchmark.c.o
Archiving .pio/build/esp32dev/libf95/liblv_examples.a
Indexing .pio/build/esp32dev/libf95/liblv_examples.a
Linking .pio/build/esp32dev/firmware.elf
.pio/build/esp32dev/libf95/liblv_examples.a(lv_demo_benchmark.c.o):(.literal.img_rgb_cb+0x0): undefined reference to `img_cogwheel_rgb'
.pio/build/esp32dev/libf95/liblv_examples.a(lv_demo_benchmark.c.o):(.literal.img_argb_cb+0x0): undefined reference to `img_cogwheel_argb'
.pio/build/esp32dev/libf95/liblv_examples.a(lv_demo_benchmark.c.o):(.literal.img_ckey_cb+0x0): undefined reference to `img_cogwheel_chroma_keyed'
.pio/build/esp32dev/libf95/liblv_examples.a(lv_demo_benchmark.c.o):(.literal.img_index_cb+0x0): undefined reference to `img_cogwheel_indexed16'
.pio/build/esp32dev/libf95/liblv_examples.a(lv_demo_benchmark.c.o):(.literal.img_alpha_cb+0x0): undefined reference to `img_cogwheel_alpha16'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1`

PC：
`/Applications/CLion.app/Contents/bin/cmake/mac/bin/cmake --build /Users/zhengruicong/Desktop/LVGL/cmake-build-debug --target LVGLRun -- -j 6
[ 12%] Built target lv_drivers
[ 40%] Built target lv_examples
[100%] Built target lvgl
Scanning dependencies of target LVGLRun
[100%] Building C object CMakeFiles/LVGLRun.dir/main.c.o
[100%] Linking C executable LVGLRun
Undefined symbols for architecture arm64:
  "_img_cogwheel_alpha16", referenced from:
      _img_alpha_cb in liblv_examples.a(lv_demo_benchmark.c.o)
  "_img_cogwheel_argb", referenced from:
      _img_argb_cb in liblv_examples.a(lv_demo_benchmark.c.o)
      _img_argb_recolor_cb in liblv_examples.a(lv_demo_benchmark.c.o)
      _img_argb_rot_cb in liblv_examples.a(lv_demo_benchmark.c.o)
      _img_argb_rot_aa_cb in liblv_examples.a(lv_demo_benchmark.c.o)
      _img_argb_zoom_cb in liblv_examples.a(lv_demo_benchmark.c.o)
      _img_argb_zoom_aa_cb in liblv_examples.a(lv_demo_benchmark.c.o)
      _sub_img_cb in liblv_examples.a(lv_demo_benchmark.c.o)
      ...
  "_img_cogwheel_chroma_keyed", referenced from:
      _img_ckey_cb in liblv_examples.a(lv_demo_benchmark.c.o)
      _img_ckey_recolor_cb in liblv_examples.a(lv_demo_benchmark.c.o)
  "_img_cogwheel_indexed16", referenced from:
      _img_index_cb in liblv_examples.a(lv_demo_benchmark.c.o)
      _img_index_recolor_cb in liblv_examples.a(lv_demo_benchmark.c.o)
  "_img_cogwheel_rgb", referenced from:
      _img_rgb_cb in liblv_examples.a(lv_demo_benchmark.c.o)
      _img_rgb_recolor_cb in liblv_examples.a(lv_demo_benchmark.c.o)
      _img_rgb_rot_cb in liblv_examples.a(lv_demo_benchmark.c.o)
      _img_rgb_rot_aa_cb in liblv_examples.a(lv_demo_benchmark.c.o)
      _img_rgb_zoom_cb in liblv_examples.a(lv_demo_benchmark.c.o)
      _img_rgb_zoom_aa_cb in liblv_examples.a(lv_demo_benchmark.c.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [LVGLRun] Error 1
make[2]: *** [CMakeFiles/LVGLRun.dir/all] Error 2
make[1]: *** [CMakeFiles/LVGLRun.dir/rule] Error 2
make: *** [LVGLRun] Error 2`

embeddedt, 2021-08-05:
Are you compiling the images in `src/lv_demo_benchmark/assets` as well?

superCong, 2021-08-05:
Copy "accesses" to LV_ demo_ After post compilation in benchmark, "fatal error:... /.. / lvgl. H: no such file or directory" appears. I have defined build in platformio compiler_ flags = -D LV_ LVGL_ H_ INCLUDE_ Simple, but it doesn't seem to work. Finally, you can only manually change.. /.. / lvgl. H to lvgl. H to compile

Copy "accesses" to LV_ demo_ After post compilation in benchmark, "fatal error:... /.. / lvgl. H: no such file or directory" appears. I have defined build in platformio compiler_ flags = -D LV_ LVGL_ H_ INCLUDE_ Simple, but it doesn't seem to work. Finally, you can only manually change.. /.. / lvgl. H to lvgl. H to compile

closed 2022-04-27 by stale[bot]