lvgl/lv_demos #119: For which version of LVGL are these demos?

SinglWolf, 2021-09-07:
As far as I understood, reading the official documentation, the ``create_cont()`` function does not exist in the current version of the LVGL, it has been canceled?

[image]

embeddedt, 2021-09-07:
These are helper functions defined [at the bottom of `lv_demo_music_main.c`](https://github.com/lvgl/lv_demos/blob/8aa1af153734fb0efe145525778f55f619a42467/src/lv_demo_music/lv_demo_music_main.c#L338).

SinglWolf, 2021-09-07:
> Это вспомогательные функции, определенные [в нижней части `lv_demo_music_main.c`](https://github.com/lvgl/lv_demos/blob/8aa1af153734fb0efe145525778f55f619a42467/src/lv_demo_music/lv_demo_music_main.c#L338).

OMG, I'm already confused about these versions, thanks.

closed 2021-09-07 by SinglWolf