lvgl/lv_demos #134: #132 brakes build

higaski, 2021-11-12:
The latest commit which merged the PR #132 broke my build. It removed font declarations needed later in the file.

```sh
lv_demo_benchmark.c:404:44: error: ‘lv_font_montserrat_12_compr_az’ undeclared (first use in this function); did you mean ‘lv_font_montserrat_12_subpx’?
  404 |     lv_style_set_text_font(&style_common, &lv_font_montserrat_12_compr_az);
      |                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                            lv_font_montserrat_12_subpx
```

liukangcc, 2021-11-14:
Sorry, I improved the code, you can review the PR.

kisvegabor, 2021-11-16:
True, see https://github.com/lvgl/lv_demos/pull/132#issuecomment-970145414

mz8023yt, 2021-12-14:
中文：这个问题需要在 lv_conf.h 文件中，将 LV_FONT_MONTSERRAT_12 宏空的值从 0 修改为 1。
English：This problem requires changing the LV_FONT_MONTSERRAT_12 macro value from 0 to 1 in the lv_conf.h file.

closed 2022-04-27 by stale[bot]