lvgl/lv_demos #141: lv_demo_widgets, no member "text_length"

epikao, 2021-11-30:
Hello, if enable lv_demo_widgets and compile, it shows dsc has no member "text_length"... 

.pio\libdeps\disco_f746ng\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c: In function 'chart_event_cb':
.pio\libdeps\disco_f746ng\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c:1308:43: error: 'lv_obj_draw_part_dsc_t' {aka 'struct <anonymous>'} has no member named 'text_length'
 1308 |                 lv_snprintf(dsc->text, dsc->text_length, "%s", month[dsc->value]);
      |                                           ^~
.pio\libdeps\disco_f746ng\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c:1311:43: error: 'lv_obj_draw_part_dsc_t' {aka 'struct <anonymous>'} has no member named 'text_length'
 1311 |                 lv_snprintf(dsc->text, dsc->text_length, "%s", month[dsc->value]);
      |                                           ^~
.pio\libdeps\disco_f746ng\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c: In function 'shop_chart_event_cb':
.pio\libdeps\disco_f746ng\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c:1423:39: error: 'lv_obj_draw_part_dsc_t' {aka 'struct <anonymous>'} has no member named 'text_length'
 1423 |             lv_snprintf(dsc->text, dsc->text_length, "%s", month[dsc->value]);
      |                                       ^~
Indexing .pio\build\disco_f746ng\lib155\libSPI.a
*** [.pio\build\disco_f746ng\lib6f7\lv_examples\lv_demo_widgets\lv_demo_widgets.c.o] Error 1

embeddedt, 2021-11-30:
Your LVGL version is outdated.

epikao, 2021-11-30:
> Your LVGL version is outdated.

Yes, but unfortunately with platformIO there is still no update available for the lvgl library. So 8.0.1 is the latest I can add to the project. See screenshot below.

[image]

embeddedt, 2021-11-30:
@kisvegabor might need to publish it manually: https://github.com/lvgl/lvgl/issues/2315#issuecomment-866099381

epikao, 2021-11-30:
> @kisvegabor might need to publish it manually: [lvgl/lvgl#2315 (comment)](https://github.com/lvgl/lvgl/issues/2315#issuecomment-866099381)

That would be really great...

kisvegabor, 2021-12-01:
I've updated all 3 repos:
- https://platformio.org/lib/show/6598/lvgl
- https://platformio.org/lib/show/6613/lv_drivers
- https://platformio.org/lib/show/13229/lv_demos (was lv_examples formerly)

Aaand, it seems this time I didn't mess it up :smile:

epikao, 2021-12-01:
> I've updated all 3 repos:
> 
> * https://platformio.org/lib/show/6598/lvgl
> * https://platformio.org/lib/show/6613/lv_drivers
> * https://platformio.org/lib/show/13229/lv_demos (was lv_examples formerly)
> 
> Aaand, it seems this time I didn't mess it up 😄

Thank you very much, I could update and compile it now without errors..
But I have still one or two issues:

```
  //lv_demo_music();
  lv_demo_widgets();
  //lv_demo_benchmark();
  //lv_demo_keypad_encoder();
  //lv_demo_stress();
```
**1st issue:**
All demos works, except lv_demo_widgets(); => it does not work, no content... 

**2th issue:**
With benchmark or keypad demo, sometimes I see wrong pixel/lines, see pictures below. What's that? 
Do I have possibly a bug with the display driver?

[image]

closed 2021-12-03 by epikao