lvgl/lv_demos #142: lv_demo_widgets, no display

epikao, 2021-12-03:
Hello

See my configuration below.
All demos works, except lv_demo_widgets. Is there maybe a bug?

Thank you

```
  //lv_demo_music(); // OK
  lv_demo_widgets(); // NOT OK
  //lv_demo_benchmark(); // OK
  //lv_demo_keypad_encoder(); // OK
  //lv_demo_stress(); // OK

[env:disco_f746ng]
platform = ststm32
board = disco_f746ng
framework = arduino
lib_deps = 
	lvgl/lv_examples@^8.1.1-dev
	lvgl/lvgl@^8.1.0
```

kisvegabor, 2021-12-03:
What if you increase `LV_MEM_SIZE`?

epikao, 2021-12-03:
change `#  define LV_MEM_SIZE (32U * 1024U)`  to `# define LV_MEM_SIZE (32U * 2048U) `  does not fix the problem.

Display difference between 1024U and 2048U:
1024U = Garbled screen
2048U = Black screen

kisvegabor, 2021-12-03:
Please enable LVGL's logging. https://docs.lvgl.io/master/porting/log.html

It also might be related: https://github.com/lvgl/lv_demos/issues/143

epikao, 2021-12-09:
> Please enable LVGL's logging. https://docs.lvgl.io/master/porting/log.html

phuu logging does not work... I don't know why, I enabled in lv_conf.h:
`#define LV_USE_LOG 1`
`#  define LV_LOG_PRINTF 0`

then in setup function:
`lv_log_register_print_cb(my_log_cb);`

then function:
```
void my_log_cb(const char * buf)
{
  Serial6.println(buf);
  Serial6.println("test");
}
```

Serial6.println("loop"); in the loop function works.

Thank you

epikao, 2021-12-10:
@kisvegabor 
> Please enable LVGL's logging. https://docs.lvgl.io/master/porting/log.html
> 
> It also might be related: #143

with all other demos I receive logging messages, but with lv_demo_widgets(); nothing happens. 
But I just saw nothing works at all with widgets, i.e. apparently the program freezes somewhere....

epikao, 2021-12-10:
could this problem have a relation to [https://github.com/lvgl/lvgl/pull/1599] ?

epikao, 2021-12-17:
> I don't think so. Why do you think it might be related?

because if activate lv_demo_widgets everything freezes/crashing...

kisvegabor, 2021-12-20:
I've just merged the demos in to the main lvgl repo. I hope it helps to solve issues like these. So, can you try the latest lvgl where everything is in one place?

epikao, 2021-12-20:
> I've just merged the demos in to the main lvgl repo. I hope it helps to solve issues like these. So, can you try the latest lvgl where everything is in one place?

Ok I see, I think this is a version that has not yet been officially released to platformIO? I'm just thinking about how I include this via platformIO cleanest (I could simply replace the lvgl folder with latest version from github or just add the new demos folder, but not sure). Current version release version according platformIO is still 8.1.0...

epikao, 2021-12-20:
> I've just merged the demos in to the main lvgl repo. I hope it helps to solve issues like these. So, can you try the latest lvgl where everything is in one place?

I replaced the lvgl folder (in platformIO) with current version from github, and include:
#include <lvgl.h>
#include <demos/lv_demos.h>

but still following compiler error:

main.cpp:(.text.setup+0x148): undefined reference to `lv_demo_widgets'

kisvegabor, 2021-12-21:
Have you enabled it in lv_conf.h? 
See https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L626

epikao, 2021-12-21:
> Have you enabled it in lv_conf.h?

YES, but I just remarked that if enabled or not, failure is the same... hmm, clean cache and so does also not help.

kisvegabor, 2021-12-22:
Ah, it's just came to my mind what could be the problem. Arduni builds only the files from the `src` folder. So you acn manually copy the `demos` folder to `src`. Probably some include path needs to be modified too.

epikao, 2021-12-22:
> Ah, it's just came to my mind what could be the problem. Arduni builds only the files from the `src` folder. So you acn manually copy the `demos` folder to `src`. Probably some include path needs to be modified too.

perfect, now it compiles without errors. But I will only be able to test next week whether the widget demo works or not, because I don't have the display hardware at hand right now. Thanks a lot, I'll get back to you.

kisvegabor, 2021-12-22:
Glad to hear that! :slightly_smiling_face:

Let's close this issue now. Please open a new one if it doesn't work.

closed 2021-12-22 by kisvegabor