lvgl/lv_demos #143: demo, wrong pixel / lines

epikao, 2021-12-03:
Hello

With benchmark or keypad demo, sometimes I see wrong pixel/lines, see pictures below. What's that? 
Do I have possibly a bug with my display driver code?

Thank you

[image]

kisvegabor, 2021-12-03:
I believe it was fixed in the latest LVGL (master branch). Can you try it out?

epikao, 2021-12-03:
> I believe it was fixed in the latest LVGL (master branch). Can you try it out?

this is my configuration:

```
[env:disco_f746ng]
platform = ststm32
board = disco_f746ng
framework = arduino
lib_deps = 
	lvgl/lv_examples@^8.1.1-dev
	lvgl/lvgl@^8.1.0
```

This is my driver:

void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )
{
  int16_t x1, y1,x2,y2;
	x1=area->x1;
	x2=area->x2;
	y1=area->y1;
	y2=area->y2;
	if (x2>479)
		x2=479;
	if (y2>271)
		y2=271;
	if (y1<0)
		y1=0;
	if (y2<0)
		y2=0;
	if (x1<0)
		x1=0;
	if (x2<0)
		x2=0;
	//lcd_fb = ( uint16_t*) (0xC0000000);
	for(int y = y1; y <= y2; y++) {
		for(int x = x1; x <= x2; x++) {
			tft.buffer[x+(y * 480)]=(color_p++->full);  //pixelColor;
		}
	}
	lv_disp_flush_ready(disp);
}

kisvegabor, 2021-12-03:
Try
```
	lvgl/lvgl#master
```

epikao, 2021-12-03:
> Try
> 
> ```
> 	lvgl/lvgl#master
> ```

I think this has the same effect as "clean all". I did it, but then I have to reset "lvgl/lvgl@^8.1.0", otherwise the conf file always changes to template.
However, still same effect, hmm

epikao, 2021-12-03:
Hello, if I use following code, see Code-1, then I have display content according picture below.

Code-1
```
  String LVGL_Arduino = "Hello World";

  // rectangle 
  
  static lv_obj_t * my_rect = lv_obj_create(lv_scr_act());
  lv_obj_set_style_radius(my_rect, 0, 0);
  lv_obj_set_pos(my_rect , 0, 0);
  lv_obj_set_size(my_rect , 480, 272);
  lv_obj_set_style_border_color(my_rect, lv_color_make(0xff,0x00,0x00), 0);
  lv_obj_set_style_border_width(my_rect, 1, 0);
  lv_obj_set_style_bg_color(my_rect , lv_palette_lighten(LV_PALETTE_GREY, 3), 0);
  

  // circle  
  static lv_obj_t * my_Cir = lv_obj_create(lv_scr_act());
  lv_obj_set_scrollbar_mode(my_Cir , LV_SCROLLBAR_MODE_OFF);
  lv_obj_set_size(my_Cir , 42, 42);
  lv_obj_set_pos(my_Cir , 270, 10);
  lv_obj_set_style_bg_color(my_Cir , lv_palette_lighten(LV_PALETTE_BLUE, 3), 0);
  lv_obj_set_style_radius(my_Cir , LV_RADIUS_CIRCLE, 0);

  // Create simple label 
  lv_obj_t *label = lv_label_create( lv_scr_act() );
  lv_label_set_text( label, LVGL_Arduino.c_str() );
  lv_obj_align( label, LV_ALIGN_CENTER, 0, 0 );
```

[image]

kisvegabor, 2021-12-08:
I'm sorry but I'm not a big expert of PlatformIO so I don't know exactly how to manage versions.
There were some related bugs in the recent versions of LVGL but they were fixed. Hopefully using the latest commit from master should solve it.

epikao, 2021-12-09:
> I'm sorry but I'm not a big expert of PlatformIO so I don't know exactly how to manage versions. There were some related bugs in the recent versions of LVGL but they were fixed. Hopefully using the latest commit from master should solve it.

Ok, I'm already using the last version, so 8.1.0, so I think this failure must be something else... difficult...

epikao, 2021-12-10:
Below the logging with my "hello world" content:

```
[Trace] (40.851, +5)     lv_indev_read_timer_cb: begin  (in lv_indev.c line #66)

[Trace] (40.862, +11)    _lv_indev_read: calling indev_read_cb  (in lv_hal_indev.c line #151)

[Trace] (40.868, +6)     lv_indev_read_timer_cb: finished       (in lv_indev.c line #119)

[Trace] (40.873, +5)     lv_timer_exec: timer callback 0x800530d finished       (in lv_timer.c line #314)

[Trace] (40.884, +11)    lv_timer_handler: finished (0 ms until the next timer call)    (in lv_timer.c line #144)

loopTest[Trace] (40.901, +17)    lv_timer_handler: begin        (in lv_timer.c line #69)

[Trace] (40.905, +4)     lv_timer_exec: calling timer callback: 0x800530d       (in lv_timer.c line #312)

[Trace] (40.915, +10)    lv_indev_read_timer_cb: begin  (in lv_indev.c line #66)

[Trace] (40.921, +6)     _lv_indev_read: calling indev_read_cb  (in lv_hal_indev.c line #151)

[Trace] (40.927, +6)     lv_indev_read_timer_cb: finished       (in lv_indev.c line #119)

[Trace] (40.938, +11)    lv_timer_exec: timer callback 0x800530d finished       (in lv_timer.c line #314)

[Trace] (40.943, +5)     lv_timer_handler: finished (0 ms until the next timer call)    (in lv_timer.c line #144)

loopTest[Trace] (40.959, +16)    lv_timer_handler: begin        (in lv_timer.c line #69)

[Trace] (40.961, +2)     lv_timer_exec: calling timer callback: 0x800530d       (in lv_timer.c line #312)

[Trace] (40.972, +11)    lv_indev_read_timer_cb: begin  (in lv_indev.c line #66)

[Trace] (40.978, +6)     _lv_indev_read: calling indev_read_cb  (in lv_hal_indev.c line #151)

[Trace] (40.989, +11)    lv_indev_read_timer_cb: finished       (in lv_indev.c line #119)

[Trace] (40.995, +6)     lv_timer_exec: timer callback 0x800530d finished       (in lv_timer.c line #314)

[Trace] (41.000, +5)     lv_timer_handler: finished (0 ms until the next timer call)    (in lv_timer.c line #144)

loopTest[Trace] (41.016, +16)    lv_timer_handler: begin        (in lv_timer.c line #69)

[Trace] (41.017, +1)     lv_timer_exec: calling timer callback: 0x800530d       (in lv_timer.c line #312)

[Trace] (41.028, +11)    lv_indev_read_timer_cb: begin  (in lv_indev.c line #66)

[Trace] (41.034, +6)     _lv_indev_read: calling indev_read_cb  (in lv_hal_indev.c line #151)

[Trace] (41.045, +11)    lv_indev_read_timer_cb: finished       (in lv_indev.c line #119)

[Trace] (41.050, +5)     lv_timer_exec: timer callback 0x800530d finished       (in lv_timer.c line #314)

[Trace] (41.062, +12)    lv_timer_handler: finished (0 ms until the next timer call)    (in lv_timer.c line #144)

loopTest[Trace] (41.072, +10)    lv_timer_handler: begin        (in lv_timer.c line #69)
```

epikao, 2021-12-15:
I think enable DMA2D could be the solution, but I have following warning:

**LVGL can't enable the clock of DMA2D**

epikao, 2021-12-15:
I think enable DMA2D could solve the problem, but I still have following warning:

#warning "LVGL can't enable the clock of DMA2D" [-Wcpp]

epikao, 2021-12-17:
I tried to activate DMA2D, see picture below the results. Unfortunately I still have this glitched rendering.
I'm using platformIO with stm32duino, LTDC 
( https://github.com/fpistm/Arduino_Core_STM32/tree/LTDC_TS_TEST_1.9.0/libraries ) and LVGL 8.1.0

[image]

epikao, 2021-12-17:
the error is simple, I forgot these two functions (in the driver function):
SCB_CleanInvalidateDCache();
SCB_InvalidateICache();

kisvegabor, 2021-12-20:
Ah, it's great that you have found it!

> now it is much better, sometimes can still see glitched rendering but very slightly....

Maybe it's just a VSYNC issue.

I close this issue then.

closed 2021-12-20 by kisvegabor