lvgl/lv_demos #149: Could you offer a simplest demo?

sonictl, 2022-01-22:
The learning curve for lvgl is sharp. Could you offer a simplest example that only contains one component in screen and compilable main.c file that out-of-box.

Thank you very much!

kisvegabor, 2022-01-22:
You can find 100+ simple examples here: https://docs.lvgl.io/master/examples.html

sonictl, 2022-01-23:
@kisvegabor Thank you very much for your soon answer!! by the way, is there any blog post/ tutorial showing how to use these simple  examples for newbie. I follows [this link](https://docs.lvgl.io/latest/en/html/get-started/quick-overview.html#quick-overview) but found their guidance is very limited. I met too many errors when trying to compile my code.

kisvegabor, 2022-01-24:
Just call e.g. `lv_example_btn_1()` in your `main` function.

sonictl, 2022-01-26:
Awsome~!

closed 2022-01-28 by sonictl