lvgl/lv_demos #152: Would a PR to update to v8.3.1 of lvgl be considered?

chmorgan, 2022-08-05:
Updated lvgl from v8.1 to 8.3.1 and seeing errors due to changes in lv_obj_draw_part_dsc_t, I think a property was renamed.

I think I can fix it, would a PR to correct be something you'd consider accepting upstream?

embeddedt, 2022-08-05:
The demos are in the main repository as of 8.2 so you shouldn't need to use this one with 8.3.

chmorgan, 2022-08-05:
@embeddedt oh very good, should I open a PR to clarify this in the README.md? It says the repo is obsolete but doesn't say when/where. Would it make sense to flag the repo as archived to be super clear that its obsolete?

kisvegabor, 2022-08-07:
[image]
It seems clear to me, but if wasn't for you, probably it isn't clear for others either. 

So please send a PR to update the top text. :slightly_smiling_face: 

As a reasonable time has passed since we merged the demos to lvgl we can archive this repo when your PR is merged.

chmorgan, 2022-08-11:
Opening a PR in just a min.

chmorgan, 2022-08-11:
https://github.com/chmorgan/lv_demos/pull/1

embeddedt, 2022-08-11:
@chmorgan I think you opened your PR on your own fork by accident.

chmorgan, 2022-08-11:
bah, https://github.com/lvgl/lv_demos/pull/153

chmorgan, 2022-08-11:
Jesus, wrong repo again...

chmorgan, 2022-08-11:
ok no, it looks ok. I thought somehow I opened against lvgl/lvgl. hah